pycom/pymakr-atom #78: Uncaught Error: watch ~/Pycom/pymakr.conf ENOSPC

blogiston, 2018-05-03:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.26.1 x64
**Electron**: 1.7.11
**OS**: Ubuntu 14.04.5
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.3.0

### Stack Trace

Uncaught Error: watch /home/david/Pycom/pymakr.conf ENOSPC

```
At fs.js:1412

Error: watch /home/david/Pycom/pymakr.conf ENOSPC
    at exports._errnoException (util.js:1050:11)
    at FSWatcher.start (fs.js:1410:19)
    at Object.fs.watch (fs.js:1436:11)
    at /packages/pymakr/lib/main/settings-wrapper.js:61:33
    at FSReqWrap.oncomplete (fs.js:114:15)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.3.0 
```

RalphHogenbirk, 2018-05-04:
Hi @blogiston, ENOSPC means there is no space on your drive, so pymakr fails to watch. I'll check if I can add nicer error handling for it.

sdaniel55, 2018-07-09:
Hi @blogiston, I'll close this issue for now. If you're still having problems, feel free to reopen it and add more details. (#54 has the same error)

closed 2018-07-09 by sdaniel55