pycom/pymakr-atom #80: pymakr

sandeepmongar, 2018-05-23:
i tried to update atom and now i could not install pymakr in atom.

RalphHogenbirk, 2018-05-23:
Hi, please provide more information so we can help you fix this problem. What error message do you get when you try to install pymakr in atom? What version of atom are you using? And which OS?

sdaniel55, 2018-07-09:
Hi @sandeepmongar, could you provide more details about your issue? (see questions above) Thanks!

FerPapi, 2018-09-19:
Hey guys, I'll take advantage of this thread: Pymakr seems to be incompatible with the latest version of Atom (1.5.1 if I'm not mistaken). Pymakr is compatible with versions 1.16 to 1.31. Any idea when pymakr can catch up? Thanks

(I tried using the plugin on VSCode and also had problems -- it simply does not work)

sdaniel55, 2018-09-19:
Hi @FerPapi 👋 

If you have any problems, please open a new issue, since your question is unrelated to this one. I'm not sure what [Atom](https://atom.io) you're using, the latest is `1.30.0`, beta is `1.31.0-beta1`, both are working fine. Please check your Atom version again, make sure you use the latest one.

FerPapi, 2018-09-19:
Hi, I will reinstall Atom then. If I have the same problem, I'll open a new issue. But I definitely have Atom 1.5.3, which is weird

[image]

FerPapi, 2018-09-19:
Sorry, I didn't realize Atom's latest is actually 1.30 and not 1.3. Somehow their version on Ubuntu's repo is not updated. 

Thx!

closed 2020-02-18 by pedroknup