pycom/pymakr-atom #81: 1.3.1 Installation Failed

daniel-rome, 2018-05-23:
Hi

I have tried to install the pymakr 1.3.1 version and it got the following error message:
```
Installing “pymakr@1.3.1” failed.Hide output…

> pymakr@1.3.1 install /tmp/apm-install-dir-118423-8499-19qwofo.1cdtxa8aor/node_modules/pymakr
> node install.js

module.js:471
    throw err;
    ^

Error: Cannot find module '/tmp/apm-install-dir-118423-8499-19qwofo.1cdtxa8aor/node_modules/pymakr/install.js'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3
npm WARN enoent ENOENT: no such file or directory, open '/tmp/apm-install-dir-118423-8499-19qwofo.1cdtxa8aor/package.json'
npm WARN apm-install-dir-118423-8499-19qwofo.1cdtxa8aor No description
npm WARN apm-install-dir-118423-8499-19qwofo.1cdtxa8aor No repository field.
npm WARN apm-install-dir-118423-8499-19qwofo.1cdtxa8aor No README data
npm WARN apm-install-dir-118423-8499-19qwofo.1cdtxa8aor No license field.
npm ERR! Linux 3.16.0-57-generic
npm ERR! argv "/opt/atom/resources/app/apm/bin/node" "/opt/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/<user>/.atom/.apm/.apmrc" "--userconfig" "/home/<user>/.atom/.apmrc" "install" "/tmp/d-118423-8499-104t48z.4vx1y06bt9/package.tgz" "--runtime=electron" "--target=1.7.11" "--arch=x64" "--global-style"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! pymakr@1.3.1 install: `node install.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the pymakr@1.3.1 install script 'node install.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the pymakr package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node install.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs pymakr
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls pymakr
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/apm-install-dir-118423-8499-19qwofo.1cdtxa8aor/npm-debug.log
```
Do you know why it could be happening? 

Installation versions below:
```
# atom -v
Atom    : 1.26.1
Electron: 1.7.11
Chrome  : 58.0.3029.110
Node    : 7.9.0
# apm -v
apm  1.19.0
npm  3.10.10
node 6.9.5 x64
atom 1.26.1
python 2.7.14
git 1.9.1
```

Thanks

RalphHogenbirk, 2018-05-23:
Hi @daniel-rome, thanks for reporting this. It's a very strange error. Looks like it can't find install.js in the root folder of the plugin (while it's definitely there). Which version of linux are you on? I've tested on linux 16.04, where I found no problem installing.

daniel-rome, 2018-05-23:
Hi, 

Just tested this on an Ubuntu 16.04 VM and it returned the same error (pymakr 1.3.2 version).
```
$ cat /proc/version
Linux version 4.13.0-41-generic (buildd@lgw01-amd64-028) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.9)) #46~16.04.1-Ubuntu SMP Thu May 3 10:06:43 UTC 2018
$ atom -v
Atom    : 1.26.1
Electron: 1.7.11
Chrome  : 58.0.3029.110
Node    : 7.9.0
$ apm -v
apm  1.19.0
npm  3.10.10
node 6.9.5 x64
atom 1.26.1
python 2.7.12
git 2.7.4
```

First test was done in a Ubuntu 14.04 LTS.

Thanks

daniel-rome, 2018-05-26:
Hi,

After cleaning all previous installation traces, I have installed the atom v1.27.1 and the the pymakr v1.3.2 package and it seems it is working properly, so probably it could be my fault.

Thank you for your help,
Daniel

closed 2018-05-26 by daniel-rome