pycom/pymakr-atom #84: Failed to activate the pymakr package on Atom 1.28.0

eriklevoll, 2018-06-22:
The module '\\?\C:\Users\Erik-Work\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 54. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

There's no version of this package that is compatible with your Atom version. The version must satisfy >= 1.16.0 <= 1.27.2

anttonij, 2018-06-23:
I'm facing the same issue on Windows running Atom 1.28.0 and trying to install pymakr 1.3.3: "There is no version of this package that is compatible with your Atom version. The version must satisfy >=1.16.0 <=1.27.2."

LoRaTheExplorer, 2018-06-24:
moi aussi

BorMi, 2018-06-25:
Same issue over here!

RalphHogenbirk, 2018-06-26:
@eriklevoll @anttonij @LoRaTheExplorer @BorMi Released Oymakr v1.3.4 which should solve this issue. Can you confirm?

eriklevoll, 2018-06-26:
I'm now getting an error:

Error: Cannot find module 'parser-byte-length'
    at Module._resolveFilename (module.js:543:15)
    at Module._resolveFilename (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\electron.asar\common\reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\module-cache.js:354:58)
    at Module.require (file:///C:/Users/Erik-Work/AppData/Local/atom/app-1.28.0/resources/app.asar/static/index.js:40:43)
    at require (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\Erik-Work\.atom\packages\pymakr\precompiles\serialport-win\lib\parsers.js:30:15)
    at Object.<anonymous> (C:\Users\Erik-Work\.atom\packages\pymakr\precompiles\serialport-win\lib\parsers.js:38:3)
    at Module.get_Module._compile (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\compile-cache.js:240:29)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///C:/Users/Erik-Work/AppData/Local/atom/app-1.28.0/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\Erik-Work\.atom\packages\pymakr\precompiles\serialport-win\lib\index.js:4:17)
    at Object.<anonymous> (C:\Users\Erik-Work\.atom\packages\pymakr\precompiles\serialport-win\lib\index.js:18:3)
    at Module.get_Module._compile (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\compile-cache.js:240:29)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///C:/Users/Erik-Work/AppData/Local/atom/app-1.28.0/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (file:///C:/Users/Erik-Work/.atom/packages/pymakr/lib/connections/pyserial.js:22:7)
    at Object.<anonymous> (file:///C:/Users/Erik-Work/.atom/packages/pymakr/lib/connections/pyserial.js:31:30)
    at Module.get_Module._compile (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\Erik-Work\AppData\Local\atom\app-1.28.0\resources\app\src\compile-cache.js:240:29)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)

RalphHogenbirk, 2018-06-26:
@eriklevoll Thanks for reporting back! I indeed see some missing packages that I've somehow missed. On my machines it's working without them. I'm going to do some tests.
In the meantime, could you try ```npm install parser-byte-length``` from the ~/.atom/packages/pymakr/ folder and see what happens?

eriklevoll, 2018-06-26:
There were indeed a few more packages missing, but after installing parser-byte-length, bindings, parser-delimiter, parser-readline, parser-ready and parser-regex it seems to be working again

RalphHogenbirk, 2018-06-26:
@eriklevoll Bindings is already in the dependencies, so that one should not have been needed. In any case I'll update the package.json to accomodate the latest serialport lib and will release the hotfix today. Thanks for helping out.

eriklevoll, 2018-06-26:
Yup, can confirm. Works nicely now!

closed 2018-06-26 by eriklevoll

steek1337, 2018-07-26:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.26.1 x64
**Electron**: 1.7.11
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.7

### Stack Trace

Failed to activate the pymakr package

```
At The module '\\?\C:\Users\Stefan\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 53. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

Error: The module '\\?\C:\Users\Stefan\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 53. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:66:33)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:76:44)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:101:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.26.1/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.26.1/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:9:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:106:36)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.2.7 
```

RalphHogenbirk, 2018-07-26:
@steek1337 please update atom and Pymakr to the latest version

Dexhub, 2019-07-25:
@steek1337 : Still an issue. I'm on the latest and greatest.

This issue happens after installing the package via atom's inbuilt installer. It is also prevalent after an install from source. 

On starting the atom app, I'm encountered with this issue.

```
The module 'repos/pymakr-atom/node_modules/bindings/build/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
Hide Stack Trace
Error: The module 'repos/pymakr-atom/node_modules/bindings/build/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:71:46)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146719)
    at bindings (repos/pymakr-atom/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (repos/pymakr-atom/precompiles/serialport-osx/lib/bindings/darwin.js:2:36)
    at Object.<anonymous> (repos/pymakr-atom/precompiles/serialport-osx/lib/bindings/darwin.js:106:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:71:46)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146719)
    at Object.<anonymous> (repos/pymakr-atom/precompiles/serialport-osx/lib/bindings/auto-detect.js:12:22)
    at Object.<anonymous> (repos/pymakr-atom/precompiles/serialport-osx/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:71:46)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146719)
    at Object.<anonymous> (repos/pymakr-atom/precompiles/serialport-osx/lib/index.js:3:17)
    at Object.<anonymous> (repos/pymakr-atom/precompiles/serialport-osx/lib/index.js:18:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:147410)
```

Mac 10.14.4
Atom version: 1.39.1
Pymakr:1.4.14

Pymakr from GitHub master branch.

Related issue which is closed: https://github.com/pycom/pymakr-atom/issues/84

Dexhub, 2019-07-26:
Got this resolved:
Had to rollback Atom 1.38.2. Pymakr seems to be having issues with the latest atom release.

[https://discuss.atom.io/t/open-how-to-prevent-atom-to-update-packages-at-each-reload/45638](https://discuss.atom.io/t/open-how-to-prevent-atom-to-update-packages-at-each-reload/45638)

groger57, 2019-07-27:
Wow. Tried x64, 32 bit, versions 1.38.2 all the way up top latest. Crashes whenever pymakr package is installed. Absolutely bad and counter productive.

jaketeater, 2019-07-29:
Also getting "Failed to activate the pymakr package"

I installed via Atom Edit->Preferences->Install...

It worked before upgrading Atom (which was rather out of date).

**Atom**: 1.39.1 x64
**Electron**: 3.1.10
**OS**: Ubuntu 18.04.2
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.14

### Stack Trace

Failed to activate the pymakr package

```
At The module '/home/jake/.atom/packages/pymakr/precompiles/serialport-linux/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Error: The module '/home/jake/.atom/packages/pymakr/precompiles/serialport-linux/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146719)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/precompiles/serialport-linux/lib/bindings/linux.js:2:36)
    at /packages/pymakr/precompiles/serialport-linux/lib/bindings/linux.js:105:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-linux/lib/bindings/auto-detect.js:16:22)
    at /packages/pymakr/precompiles/serialport-linux/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-linux/lib/index.js:3:17)
    at /packages/pymakr/precompiles/serialport-linux/lib/index.js:18:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147410)
```

jdcrunchman, 2019-07-30:
After a huge marathon work session,  dealing with poor vision,  and not able to read the console,   my partner and I tried different earlier versions of Atom.   We found that Atom ver 1.38.2 seemed to work with the latest PyMakr.    I had installed it more then 2 months ago,  apparently an earlier version,  and people were talking about only using Python 2.7,  but was unable to figure out how to tell Atom to use Python 2.7,  on a whim,   After downloading Atom,  and then running it.   Strange,  it already knew I was using PyMakr,  and the console showed up in its tiny hard to read font,   so I did a download and all seemed to be loaded into the ESP32,  so I hope this helps others.

i really wanted to get "rdagger68" into our Zoom video conferencing room,  but he didn't seem at all interested....  Oh well....

John

MJtuks, 2019-07-31:
After opening Atom I get this error:

`[Enter steps to reproduce:]

1. Open Atom

**Atom**: 1.39.1 x64
**Electron**: 3.1.10
**OS**: Microsoft Windows 10 Home Single Language
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.14

### Stack Trace

Failed to activate the pymakr package

```
At The module '\\?\C:\Users\mjaa\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Error: The module '\\?\C:\Users\mjaa\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:146719)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:3:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:18:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:147410)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.4.14 
```

`