pycom/pymakr-atom #85: Failed to load the pymakr package

ClarkP57, 2018-06-27:
This seems to be a problem with the 32 bit version of atom. The same problem exists on two different machines running the 32 bit version. 

The error was thrown from the pymakr package. This issue has already been reported. 
(Failed to load the pymakr package  #53)
 I've tried the relevant solutions offered but to no avail.

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.27.2 ia32
**Electron**: 1.7.15
**OS**: Microsoft Windows 7 Starter 
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.3.5

### Stack Trace

Failed to load the pymakr package

```
At The module '\\?\C:\Users\ClarkP\.atom\packages\pymakr\precompiles\serialport-win32\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

Error: The module '\\?\C:\Users\ClarkP\.atom\packages\pymakr\precompiles\serialport-win32\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:66:33)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/precompiles/serialport-win32/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win32/lib/bindings/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.27.2/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win32/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win32/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.27.2/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win32/lib/index.js:3:17)
    at /packages/pymakr/precompiles/serialport-win32/lib/index.js:18:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:106:36)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.3.5 
```

AndreasBruring, 2018-06-28:
Same here

i am trying to revert to atom 1.27.2 when i got this error
because the hotfix seems to create BSOD's in version 1.28 of atom
something related to the drivers when i am trying code in the REPL

Its been trouble the whole week with the new atom update

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.27.2 x64
**Electron**: 1.7.15
**OS**: Microsoft Windows 10 Home
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.3.5

### Stack Trace

Failed to load the pymakr package

```
At The module '\\?\C:\Users\bruring\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

Error: The module '\\?\C:\Users\bruring\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:66:33)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.27.2/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.27.2/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:3:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:18:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.27.2/resources/app/src/native-compile-cache.js:106:36)
```

### Commands

```
     -9:21.5.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
busy-signal 1.4.3 
intentions 1.1.5 
linter 2.2.0 
linter-python 3.1.2 
linter-ui-default 1.7.1 
pymakr 1.3.5 
```

hallard, 2018-07-08:
Any hints ? same issue here with 1.27.2 x64 Windows 10, tried with pymakr 1.34 1.3.5 and 1.4.0

RalphHogenbirk, 2018-07-09:
@hallard please update Atom to v1.28 and Pymakr to v1.4.0. The serialport library has been pre-compiled for the nodejs version that atom 1.28 is using. Unfortunately that breaks compatibility with any lower versions.

Let me know if if still have any issues after that.

sdaniel55, 2018-07-09:
Hi @ClarkP57 👋 
Did you manage to fix this? Newer version of Atom (1.2.8) and pymakr (1.4.0) is available. Let me know if you're still having problems!

hallard, 2018-07-09:
@RalphHogenbirk that's what I've done to be sure (was in 1.27 lastest) and issued #89 
I'm at work on another computer nodejs v8.11.1 atom 1.28.1 and I'm re-installing pymakr V1.4.0
all went fine on this one

RalphHogenbirk, 2018-07-09:
Ok good to hear. Updating to 1.28.0 should definitely fix it. If your pc at home says 1.27 is latests, download the latest manually from https://github.com/atom/atom/releases

ClarkP57, 2018-07-09:
1.28.0 solved my problem.

closed 2018-07-09 by sdaniel55

steek1337, 2018-07-26:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.26.1 x64
**Electron**: 1.7.11
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.1

### Stack Trace

Failed to load the pymakr package

```
At The module '\\?\C:\Users\Stefan\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

Error: The module '\\?\C:\Users\Stefan\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:66:33)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.26.1/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.26.1/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:3:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:18:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:106:36)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.4.1 
```

RalphHogenbirk, 2018-07-26:
@steek1337 the solution is already posted. Please update your Atom

justb4, 2019-01-30:
Encountered (and solved) this problem in following version config:

* Atom 1.34.0 x64 - Mac OSX 10.11.6 El Capitan
* pymakr 1.4.10

Error message: _"Failed to load the pymakr package"_. A link in the Atom warning panel lead me to this issue.

In my case the problem was that `pymakr` could not locate the file `serialport.node` in a series of directories under `packages/pymakr/node_modules/serialport`. I found `serialport.node`  from the install under `packages/pymakr/build`. After copying to `packages/pymakr/node_modules/serialport/build/serialport.node` the problem was gone, but IMHO a more solid solution is needed.

RalphHogenbirk, 2019-01-30:
@justb4 thanks for the report. Interesting problem, because if anything goes wrong with the serialport lib inside the node_module folder, Pymakr should fall back to the precompiled version of serialport (in precompiles/serialport-osx in your case). I'm curious why that didn't work in your case. Could you try to reproduce the problem once more by temporarily removing node_modules/serialport/build/serialport.node again and posting the stacktrace that comes with the resulting 'failed to load the pymakr package'? That would help me a lot in understanding the issue.

AndrewStyan, 2019-02-08:
Hi @RalphHogenbirk , I had the same problem as @justb4 when attempting to install pymakr.
It worked when I copied serialport.node into the folder packages/pymakr/node_modules/serialport/build (which I I had to create).
Andrew

stacktrace:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.34.0 x64
**Electron**: 2.0.16
**OS**: Mac OS X 10.14.2
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.10

### Stack Trace

Failed to load the pymakr package

```
At Could not locate the bindings file. Tried:
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/build/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/build/Debug/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/build/Release/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/out/Debug/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/Debug/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/out/Release/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/Release/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/build/default/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/compiled/8.9.3/darwin/x64/serialport.node

Error: Could not locate the bindings file. Tried:
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/build/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/build/Debug/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/build/Release/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/out/Debug/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/Debug/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/out/Release/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/Release/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/build/default/serialport.node
 → /Users/Andrew/.atom/packages/pymakr/node_modules/serialport/compiled/8.9.3/darwin/x64/serialport.node
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:93:9)
    at /packages/pymakr/node_modules/serialport/lib/bindings/darwin.js:2:36)
    at /packages/pymakr/node_modules/serialport/lib/bindings/darwin.js:106:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146684)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150231)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:145974)
    at /packages/pymakr/node_modules/serialport/lib/bindings/auto-detect.js:12:22)
    at /packages/pymakr/node_modules/serialport/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146684)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150231)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:145974)
    at /packages/pymakr/node_modules/serialport/lib/index.js:3:17)
    at /packages/pymakr/node_modules/serialport/lib/index.js:18:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146684)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150231)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:145974)
    at /packages/pymakr/lib/connections/pyserial.js:24:18)
    at /packages/pymakr/lib/connections/pyserial.js:30:30)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.4.10 
```

linucks, 2019-04-09:
I've just encountered this with Atom 1.35.1 x64 on OSX 10.14 with pymakr 1.4.12

Could not locate the bindings file. Tried:
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/build/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/build/Debug/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/build/Release/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/out/Debug/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/Debug/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/out/Release/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/Release/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/build/default/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/compiled/8.9.3/darwin/x64/serialport.node
Hide Stack Trace
Error: Could not locate the bindings file. Tried:
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/build/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/build/Debug/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/build/Release/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/out/Debug/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/Debug/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/out/Release/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/Release/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/build/default/serialport.node
 → /Users/jmht/.atom/packages/pymakr/node_modules/serialport/compiled/8.9.3/darwin/x64/serialport.node
    at bindings (/Users/jmht/.atom/packages/pymakr/node_modules/bindings/bindings.js:93:9)
    at Object.<anonymous> (/Users/jmht/.atom/packages/pymakr/node_modules/serialport/lib/bindings/darwin.js:2:36)
    at Object.<anonymous> (/Users/jmht/.atom/packages/pymakr/node_modules/serialport/lib/bindings/darwin.js:106:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146684)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150231)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:145974)
    at Object.<anonymous> (/Users/jmht/.atom/packages/pymakr/node_modules/serialport/lib/bindings/auto-detect.js:12:22)
    at Object.<anonymous> (/Users/jmht/.atom/packages/pymakr/node_modules/serialport/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146684)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150231)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:145974)
    at Object.<anonymous> (/Users/jmht/.atom/packages/pymakr/node_modules/serialport/lib/index.js:3:17)
    at Object.<anonymous> (/Users/jmht/.atom/packages/pymakr/node_modules/serialport/lib/index.js:18:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146684)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150231)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:145974)
    at Object.<anonymous> (/Users/jmht/.atom/packages/pymakr/lib/connections/pyserial.js:24:18)
    at Object.<anonymous> (/Users/jmht/.atom/packages/pymakr/lib/connections/pyserial.js:30:30)

I was able to fix it by creating the following directory:
mkdir ~/.atom/packages/pymakr/node_modules/serialport/build

and copying a serialport.node file I found into it:
cp  ~/.atom/packages/pymakr/build/serialport.node ~/.atom/packages/pymakr/node_modules/serialport/build

krisavi, 2019-06-26:
Seems the issue is back, copying the file does not help.

Atom 1.39.0-beta2
Pymakr 1.4.14

The module '\\?\C:\Users\Mercor\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
Hide Stack Trace
Error: The module '\\?\C:\Users\Mercor\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///C:/Users/Mercor/AppData/Local/atom/app-1.39.0-beta2/resources/app.asar/static/index.js:71:46)
    at require (C:\Users\Mercor\AppData\Local\atom\app-1.39.0-beta2\resources\app\static\<embedded>:11:146719)
    at bindings (C:\Users\Mercor\.atom\packages\pymakr\node_modules\bindings\bindings.js:81:44)
    at Object.<anonymous> (C:\Users\Mercor\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings\win32.js:2:36)
    at Object.<anonymous> (C:\Users\Mercor\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings\win32.js:111:3)
    at Module.get_Module._compile (C:\Users\Mercor\AppData\Local\atom\app-1.39.0-beta2\resources\app\static\<embedded>:11:147410)
    at Object.value [as .js] (C:\Users\Mercor\AppData\Local\atom\app-1.39.0-beta2\resources\app\static\<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///C:/Users/Mercor/AppData/Local/atom/app-1.39.0-beta2/resources/app.asar/static/index.js:71:46)
    at require (C:\Users\Mercor\AppData\Local\atom\app-1.39.0-beta2\resources\app\static\<embedded>:11:146719)
    at Object.<anonymous> (C:\Users\Mercor\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings\auto-detect.js:8:22)
    at Object.<anonymous> (C:\Users\Mercor\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings\auto-detect.js:19:3)
    at Module.get_Module._compile (C:\Users\Mercor\AppData\Local\atom\app-1.39.0-beta2\resources\app\static\<embedded>:11:147410)
    at Object.value [as .js] (C:\Users\Mercor\AppData\Local\atom\app-1.39.0-beta2\resources\app\static\<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///C:/Users/Mercor/AppData/Local/atom/app-1.39.0-beta2/resources/app.asar/static/index.js:71:46)
    at require (C:\Users\Mercor\AppData\Local\atom\app-1.39.0-beta2\resources\app\static\<embedded>:11:146719)
    at Object.<anonymous> (C:\Users\Mercor\.atom\packages\pymakr\precompiles\serialport-win\lib\index.js:3:17)
    at Object.<anonymous> (C:\Users\Mercor\.atom\packages\pymakr\precompiles\serialport-win\lib\index.js:18:3)
    at Module.get_Module._compile (C:\Users\Mercor\AppData\Local\atom\app-1.39.0-beta2\resources\app\static\<embedded>:11:147410)

RalphHogenbirk, 2019-06-27:
@krisavi 1.39 beta uses a different version of node than the stable release. If you install Pymakr using the stable release and then try to use it in the beta release, you get this issue. Quickest solution is to uninstall and install again using the beta release.

krisavi, 2019-06-27:
@RalphHogenbirk Thanks, I just opened my atom and it upgraded itself from older beta to newer. Installed non-beta version instead of beta.

ClarkP57, 2019-07-23:
Here we go again.
Any suggestions would be greatly appreciated.

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.39.0 x64
**Electron**: 3.1.10
**OS**: Microsoft Windows 10 Enterprise
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.14

### Stack Trace

Failed to load the pymakr package

```
At The module '\\?\C:\Users\papkecs\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Error: The module '\\?\C:\Users\papkecs\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:146719)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:3:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:18:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:147410)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.4.14 
```

Caterina1996, 2019-07-24:
Hi guys!

I've got a similar problem:

he module '\\?\C:\Users\VicenÃ§ Salas\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
Hide Stack Trace
Error: The module '\\?\C:\Users\VicenÃ§ Salas\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///C:/Users/Vicen%C3%A7%20Salas/AppData/Local/atom/app-1.39.0/resources/app.asar/static/index.js:71:46)
    at require (C:\Users\Vicenç Salas\AppData\Local\atom\app-1.39.0\resources\app\static\<embedded>:11:146719)
    at bindings (C:\Users\Vicenç Salas\.atom\packages\pymakr\node_modules\bindings\bindings.js:81:44)
    at Object.<anonymous> (C:\Users\Vicenç Salas\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings\win32.js:2:36)
    at Object.<anonymous> (C:\Users\Vicenç Salas\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings\win32.js:111:3)
    at Module.get_Module._compile (C:\Users\Vicenç Salas\AppData\Local\atom\app-1.39.0\resources\app\static\<embedded>:11:147410)
    at Object.value [as .js] (C:\Users\Vicenç Salas\AppData\Local\atom\app-1.39.0\resources\app\static\<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///C:/Users/Vicen%C3%A7%20Salas/AppData/Local/atom/app-1.39.0/resources/app.asar/static/index.js:71:46)
    at require (C:\Users\Vicenç Salas\AppData\Local\atom\app-1.39.0\resources\app\static\<embedded>:11:146719)
    at Object.<anonymous> (C:\Users\Vicenç Salas\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings\auto-detect.js:8:22)
    at Object.<anonymous> (C:\Users\Vicenç Salas\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings\auto-detect.js:19:3)
    at Module.get_Module._compile (C:\Users\Vicenç Salas\AppData\Local\atom\app-1.39.0\resources\app\static\<embedded>:11:147410)
    at Object.value [as .js] (C:\Users\Vicenç Salas\AppData\Local\atom\app-1.39.0\resources\app\static\<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///C:/Users/Vicen%C3%A7%20Salas/AppData/Local/atom/app-1.39.0/resources/app.asar/static/index.js:71:46)
    at require (C:\Users\Vicenç Salas\AppData\Local\atom\app-1.39.0\resources\app\static\<embedded>:11:146719)
    at Object.<anonymous> (C:\Users\Vicenç Salas\.atom\packages\pymakr\precompiles\serialport-win\lib\index.js:3:17)
    at Object.<anonymous> (C:\Users\Vicenç Salas\.atom\packages\pymakr\precompiles\serialport-win\lib\index.js:18:3)
    at Module.get_Module._compile (C:\Users\Vicenç Salas\AppData\Local\atom\app-1.39.0\resources\app\static\<embedded>:11:147410)

This happen just when I restarted atom and it automatically upgraded.

Do you have any idea of how to solve this?

Thanks!