pycom/pymakr-atom #86: size of panel

tooqui, 2018-06-28:
Hi all,

I am starting to work with atom as I was using plain text editors for my small projects. 
Situation is after installing latest atom and the Pymakr plugin it all works fine. except that the pane for the REPL window can not be resized further than a row wich makes it unusable.
[image]
Situation is I can make it even narrower but when I click on the panel to make it higher only allows me to grow to the size on the picture. 

Is there any conf parameter I am missing?

Thank you for your help

tooqui, 2018-06-28:
somehow I just rebooted the computer and now works fine....

closed 2018-06-28 by tooqui

RalphHogenbirk, 2018-07-02:
@tooqui I had one other report of this. Still don't know what causes it. Either it's some edge case usage of the buttons / resizing the terminal. Or it's another plugin that causes issues with the layout. let me know if you encounter this issue again and have an idea what steps cause it.

tooqui, 2018-07-02:
@RalphHogenbirk I just disabled the other 3 packages installed. Rebooted Atom and Pymakr sterted to work as it is supposed.
Then I reenabled and so far as of today the issue disapeared.

RalphHogenbirk, 2018-07-02:
@tooqui Ok, can you tell me which 3 packages just for the record? Maybe I can see a pattern when someone else has this issue as well.

tooqui, 2018-07-02:
@RalphHogenbirk ops forgot to attach...
[image]

jamesbperry, 2018-09-09:
I just had the same issue with Atom 1.30.0 and Pymakr 1.4.2. The workaround in my case was to disable atom-ide-ui, which happens to be the only other non-core package that was installed. 

Kind of an odd welcome to Pymakr, since the Atom adventure began due to the VSCode version being [broken](https://forum.pycom.io/topic/3657/lost-pymakr-pycom-console-in-vscode-after-update-vscode).

cristhoper, 2018-10-01:
i disable `nuclide` and it works.