pycom/pymakr-atom #88: Run button drops comms to Pycom devices on second use

prawnhead, 2018-07-02:
Issue:
When stopping running code on a device using the Run button, communications to the device are lost. This is not intermittent, it occurs every time the Run button is used to stop running code.

Steps to reproduce:
1. When the Pymakr Run button is used for the first time, the name of the code file run is printed to the REPL and the code is run on the device.
2. **On second press** of the Run button, REPL ceases printing output from the device as expected **but the REPL prompt does not return**
3. A third press of the Run button prints the name of the code file to REPL but no output from the device.
4. Subsequent presses of the Run button appear to have no effect.
5. Pressing reset on the device does not return the REPL prompt
6. Clicking Disconnect produces the message "> Failed to connect (Error: Port is not open). Click here to try again." and buttons Upload, Download and Run disappear.
7. Clicking Connect returns to the REPL prompt, but Upload, Download and Run do not appear.
8. Closing and reopening of Atom IDE is required to make Upload, Download and Run buttons reappear.

Versions:
- Atom 1.28.0 x64 with pymakr 1.3.5 on Win10 x64
- LoPy 1.0: (sysname='LoPy', nodename='LoPy', release='1.18.0', version='v1.8.6-849-046b350 on 2018-06-01', machine='LoPy with ESP32', lorawan='1.0.2')
- LoPy4: (sysname='LoPy4', nodename='LoPy4', release='1.18.0', version='v1.8.6-849-046b350 on 2018-06-01', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

Code to reproduce (any code will show fault)
```
import time

while(True):
    print("Test")
    time.sleep(1)
```

prawnhead, 2018-07-02:
Workaround:
Stop running code on the device using Ctrl-C

sdaniel55, 2018-07-09:
Hi @prawnhead, could you update pymakr to `1.4.0` and try it again? Thanks!

prawnhead, 2018-07-19:
Sorry for the delay. Been on holidays :P. I have updated to Atom 1.28.2 (x64) with Pymakr 1.4.1. Behaviour is exactly as reported under Steps to Reproduce in my original post. Let me know if I can provide more info. Thanks, Chris.

prawnhead, 2018-07-26:
Just checking back in. Atom 1.28.2 and Pymakr 1.4.1 are still current and the issue is not yet resolved as far as I can test.

sdaniel55, 2018-07-27:
Hi @prawnhead, I tried reproducing your issue, however I couldn't reach point 4 (although I'm using macOS). I believe when you run the code, the first press runs it, second press stops it. The REPL console always returned after the second press for me.

As for the disappearing buttons, have you tried closing the pymakr plugin a few times (with the arrow on the right hand side)? Sometimes that works for me.

RalphHogenbirk, 2018-10-01:
@prawnhead has this issue been solved in the meantime? Or is it still active?

M0GQB, 2018-10-02:
I get very similar problems running my code reading from both the PyTrack and the expansion board 3.0.  Using the "Packages/Pymakr/Toggle Pycom Console" to close and re-open the console window seems to restore the lost options of Upload, Download and Run.  However, using Ctrl-C to stop running code doesn't always work.  I'm using Atom 1.28.2, not sure about the Pymakr version - how do I check?

krismc, 2019-04-19:
I found Packages/Pymakr/Run the current file restores Upload/Download/Run buttons that often vanish after a reconnect. Heaps quicker than quitting/reloading the package.

Stayfire, 2019-06-05:
Hi, I have the same problem, when I try to run or upload I receive the message: failed to connect. click here to try again.

I'm using a Wipy card with a pysense expansion card and I'm running Windows 7 pro, I've assigned the card to the COM8 port and upgraded the firmware. Atom sometimes detects pycom but when I want to upload it tells me: Error: Writing to COM port (GetOverlappedResult): Operation aborted)

Versions:
              Atom: 1.37.0 x64
              Pymakr: 1.4.13
And I tried with 
              Pymakr: 1.0.1, always the same problem.

Moreover, if I run, then upload a cancel button appears but I can't click on it and connect, upload and download buttons disappear.
I need help to solve that please.

pedroknup, 2020-02-18:
Fixed on Pymakr 2.0

closed 2020-02-18 by pedroknup