pycom/pymakr-atom #89: Could not locate the bindings file

hallard, 2018-07-08:
After experiencing the #85 I tried updating node js to v8.11.13 and atom to v 1.28.1 but looks like it does not work

Install fail with this error code

[image]

Any hints ? I just eed to update firmware to customer device and just spent 1H30 to get things working, it was working fine until I updated to 1.4.0 today, now nothings works ;-(

hallard, 2018-07-08:
Dirty hack to solve the issue copied folder `build` from pymakr 
`C:\Users\Charles\.atom\packages\pymakr\precompiles\serialport-win`
to
`C:\Users\Charles\node_modules\bindings\`

RalphHogenbirk, 2018-07-09:
Hi @hallard, that is a strange bug. It shouldn't search in the C:\Users\Charles\node_modules folder for those bindings. 

Your workaround is good, but I'm curious what causes this issue. Can you check if the serialport.node file is present in this folder? ```C:\Users\Charles\.atom\packages\pymakr\node_modules\bindings\build```

Let me know!

hallard, 2018-07-09:
I'll check this one at home tonight

hallard, 2018-07-10:
yes it's there

[image]

RalphHogenbirk, 2018-07-11:
Thanks for sharing that info. It's strange, that is one of the places it normally looks. I have no idea why it doesn't find it, and searches in your home directory for it. 
I realise I'm still on 1.28.0 so I'll update to the latest bugfix release and will test it on windows, see if it's that.
One other thing it could be: I've see one or two cases in the lifetime of the Pymakr plugin where the plugin wouldn't update properly on windows. It skips updating the serialport libs in the /precompiles folder, maybe because this is somehow still in use. Fix for that would be to manually remove the complete pymakr folder from ```~/.atom/packages/``` and re-install the package.

RalphHogenbirk, 2018-10-01:
This issue should be solved with the latest versions, so i'm closing this now. Feel free to re-open or create a new issue when needed.

closed 2018-10-01 by RalphHogenbirk