pycom/pymakr-atom #92: Statusbar is not resizable

barryknobel, 2018-08-15:
The statusbar at the bottom of the screen is not resizable.

[image]

RalphHogenbirk, 2018-08-27:
@barryknobel Thanks for reporting, which OS and which Atom/Pymakr versions are you using? I've seen this before, but only on linux, and I haven't been able to reproduce it.

lemariva, 2018-09-10:
I have/had the same problem. Atom: 1.30 Pymakr: 1.4.2 OS: Windows 10
Solution: if you disable the atom-ide-ui package, the console goes back to the right size.

tmmorin, 2019-03-17:
I have the same issue (pymakr  1.4.11, atom 1.35.1 on Linux).
The same workaround works for me as well.
It would be great to have pymakr work out of the box, though...

stevempotter, 2019-03-18:
@RalphHogenbirk, 
I was frustrated by this bug also. I am using Mac 10.13.6 and Atom 1.35.1 and pymakr package 1.4.12.
I note that I had to res
[image]
tart Atom after disabling the atom-ide-ui package in order to be able to see more than one line of my connection pane.

pedroknup, 2020-02-18:
Fixed on Pymakr 2.0

closed 2020-02-18 by pedroknup