pycom/pymakr-atom #94: Autoconnect on USB in Pymakr is not working for windows.

Aditya18Jain, 2018-09-06:
Pr-requisite: 
Port is open.
 
Steps:
1. Check 'Autoconnect on USB' checkbox.
2. Restart Pymakr and connect device.

Result:
It says "No Pycom board is found on USB" but when i do get serial port it says 1 serial port found.
Refer attached screenshot.

Note: Please let me know for additional information (I work for Pycom remotely)
[image]

RalphHogenbirk, 2018-09-10:
Hi @Aditya18Jain, is this a prototype board, or a board connected through your own FTDI converter? Right now, the autoconnect only connects to boards that have the comport manufacturer set as Pycom, Pycom Ltd. or FTDI (which covers all our current and old boards).

Aditya18Jain, 2018-09-15:
Hi @RalphHogenbirk ,
I got board from Pycom. So i really don't know the manufacturer. It was sent by Pycom Ltd.

jirikrepl, 2018-09-17:
Hello,
I tried to connect same board firstly to Atom on MacOS and then on Windows.

On MacOS after `get serial ports` board is recognized as Pycom and auto-connect works well.
[image]

But on Windows same board after `get serial ports` is connected to COM3 and has a manufacturer Microsoft. Then auto-connect doesn't work there. (I had enabled auto-connect, but I don't have opened global settings page on second screenshot)

[image]

If I set COM3 and disable auto-connect board is connected well.

[image]

jirikrepl, 2018-09-17:
I used Expansion board 3 + FiPy

RalphHogenbirk, 2018-09-17:
I suspect this happens with the expansionboard 3 only, which shows up as a 'Microsoft' board on windows machines. I'll test this tomorrow when I have one available. If true, I'll add this to the list of valid manufacturers and it should work with autoconnect.

RalphHogenbirk, 2018-10-01:
The latest version of the plugin has a fix for this, which allows it to autoconnect to boards returning `Microsoft` or `Microchip Technology, Inc.` as manufacturer name. If you still have issues with the feature feel free to re-open the issue.

closed 2018-10-01 by RalphHogenbirk

Milan991, 2019-02-14:
@RalphHogenbirk This issue is still persistent.
Extension Board 3 + Lopy4
OS: Windows 10
IDE: Visual Studio Code

sabbir4455, 2019-03-23:
Hi...I'm using fipy with the pycom extension board V3.1 in windows7. I have successfully installed driver and upgraded the latest firmware 1.15.1. Also installed Atom with the pymakr 1.4.12. I tried with both connecting via USB and telnet and got the same error.

AutoConnect enabled, ignoring 'address' setting (see Global Settings)
Searching for PyCom boards on serial...
No PyCom boards found on USB

Found 2 serialports
COM11 (Pycom Ltd) (copied to clipboard)
COM4 (Intel)
AutoConnect: No device available
AutoConnect: No device available

Someone please help.......is it related to pymakr library issue? Also my REPL mode is not working while try to write or paste anything.

[image]

RalphHogenbirk, 2019-03-25:
@sabbir4455 looks like the manufacturer of your board is slightly different than what Pymakr expects. Your board displays as 'Pycom Ltd' where Pymakr looks for 'Pymakr Ltd.' (with punctuation point). I'd suggest adding the none punctuated one to the list of 'autoconnect comport manufacturers' that you'll find at the bottom of the global settings. After that, autoconnect should work.