pycom/pymakr-atom #95: Pymakr not Connecting

hypePG, 2018-09-19:
Hey,

I am trying to connect to a SiPy on a Pysense Board using Atom 1.30.0.
I installed Pymakr 1.4.2 and configured it according to the manual.

If I am listing the serial ports, PyMakr is correctly identifying this device:

/dev/ttyACM0

But hitting the connect button only brings me this far:

Connecting on /dev/ttyACM0...

Nothing happens. I am running Ubuntu 18.04.
In the readme I read, that there where problems with python3.6 so I reinstalled pymakr afer using the workaround which is telling apm to use python2.7.

The SiPy is alive with a blue led flashing every 4 seconds.

Please help.

Thanks in advance.

hypePG, 2018-09-20:
In the meantime i tested it with another SiPy which is also not working. I also tried to change the loglevel to "silly" to get some more information about what could be the cause for this behaviour. But there isnt much more logged during the connection atempt.

I also delete and installed Atom again. But there are still some files and properties cached, because when I install the plugin again, the device adress ist still saved. I deleted the settings folder under /home/user./atom aswell.

I also tried VSCode on my machine, but in this case the plugin isn't even shown after the install.

RalphHogenbirk, 2018-09-20:
HI @hypePG, 
When it says 'Connecting on /dev/ttyACM0', does it show 'connected' and a green V on the left top of the terminal? or still 'disconnected' and a red X? It could be that it does connect, but the device is not outputting anything because you have some code running on it. In that case after it connects, you could do ctrl-c to stop the code, or even ctrl-f to force a safe boot.

If that's not the case, make sure that there is no other program or atom window connected to the device by accident, the device only supports one connection at a time.

If that all fails, try to connect to the device using the 'screen' command in the linux terminal: ```screen /dev/ttyACM0 115200``` If that connects, we know that it might be an atom issue. If it doesn't, it's likely a driver or hardware issue.

hypePG, 2018-09-20:
Hi @RalphHogenbirk,

thanks for your answer. It states "disconnected" and a red x. My button presses have no influence on the the connection process. The device is factory new and despite the firmware update, there were no changes made.

There is no other program running which could connect to the device.

Thanks for the hint! The connection via screen works and I can execute commands on the board.
Any chance to get a deeper look insight Atom?

Regards,
Max

hypePG, 2018-09-21:
Hi @RalphHogenbirk ,
I was curious and investigated a little deeper.... And I finally managed to get it working.

The first hint was your information about the "screen" command. To successfully connect to the device I needed a "sudo screen". This made me think about the permissions of Atom. In my early attempts I tried to install and run every component (Atom, Pymakr Plugin) as root (in a sandbox) with no success.

I than read somewhere online, that there is the "dialout" group which is permitting the user to open serial connections.

The missing link was:

`sudo usermod -a -G dialout $USER`

Since that it is working like a charm. It is a pitty that the documentation doesn't give a better help for problems like that. I can't believe that I am the only one experiencing this problem since Atom and the plugin should never run with root privileges.

Whatever, thanks for your help!

RalphHogenbirk, 2018-09-21:
HI @hypePG , glad you figured it out! I've seen this before, a long time ago, but I guess it never made it's way into the docs. I've not seen it since, so I guess most people already run atom with root privileges on Linux? I'm not sure. In any case I'll add the dialout command to the readme file under 'common issues. So thanks for that info!

closed 2018-09-21 by RalphHogenbirk

PDWest, 2020-11-22:
I've got this problem w/a new install of linux mint 20, with a new Atom install on top of that and a new Pymakr on install on that.  
The screen command (w/sudo) from above works, but no luck with the 'sudo usermod -a -G dialout $USER' idea.

Is there any updated info on this??

sanjaysharathi, 2021-05-02:
In