pycom/pymakr-atom #98: Pymakr plugin won't install on ATOM

rossdem699, 2018-09-28:
I have tried to install the pymakr plugin in ATOM and i can't get it to install.
This is the error i get - 

Installing “pymakr@1.4.3” failed.Hide output…

> serialport@6.2.2 install /tmp/apm-install-dir-118828-5706-yoszgy.3jnxt7qfr/node_modules/pymakr/node_modules/serialport
> prebuild-install || node-gyp rebuild

/tmp/apm-install-dir-118828-5706-yoszgy.3jnxt7qfr
└── (empty)

npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
sh: 1: prebuild-install: Permission denied
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/tmp/apm-install-dir-118828-5706-yoszgy.3jnxt7qfr/node_modules/pymakr/node_modules/serialport/build'
gyp ERR! stack     at Error (native)
gyp ERR! System Linux 4.4.0-131-generic
gyp ERR! command "/opt/atom/resources/app/apm/bin/node" "/opt/atom/resources/app/apm/node_modules/.bin/node-gyp" "rebuild"
gyp ERR! cwd /tmp/apm-install-dir-118828-5706-yoszgy.3jnxt7qfr/node_modules/pymakr/node_modules/serialport
gyp ERR! node -v v6.9.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/tmp/apm-install-dir-118828-5706-yoszgy.3jnxt7qfr/package.json'
npm WARN apm-install-dir-118828-5706-yoszgy.3jnxt7qfr No description
npm WARN apm-install-dir-118828-5706-yoszgy.3jnxt7qfr No repository field.
npm WARN apm-install-dir-118828-5706-yoszgy.3jnxt7qfr No README data
npm WARN apm-install-dir-118828-5706-yoszgy.3jnxt7qfr No license field.
npm ERR! Linux 4.4.0-131-generic
npm ERR! argv "/opt/atom/resources/app/apm/bin/node" "/opt/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/ross/.atom/.apm/.apmrc" "--userconfig" "/home/ross/.atom/.apmrc" "install" "/tmp/d-118828-5706-10e49ty.p73wcgcik9/package.tgz" "--runtime=electron" "--target=1.7.11" "--arch=x64" "--global-style"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.2.2 install script 'prebuild-install || node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     prebuild-install || node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/apm-install-dir-118828-5706-yoszgy.3jnxt7qfr/npm-debug.log
npm ERR! code 1

i have run the following command as i am running python 3.6.4 but i have also installed python 2.7.
ross@ross-TravelMate-5335 ~ $ sudo echo "python=/usr/bin/python2.7" >> ~/.atom/.apmrc
ross@ross-TravelMate-5335 ~ $ apm --versionapm  1.19.0
npm  3.10.10
node 6.9.5 x64
atom 1.26.1
python 3.6.4
git 1.9.1

if the python 3.6.4 is the issue - how do i change it to python 2.7 as the command above isn't working?

RalphHogenbirk, 2018-10-01:
Hi @rossdem699, strange that the command isn't working. Could you check the `~/.atom/.apmrc` file manually and see if has the line 'python=/usr/bin/python2.7'?

RalphHogenbirk, 2018-10-01:
I've found the issue and it was with the package.json. I've released an update now with pymakr v1.4.4. Let me know if that solves your issue

rossdem699, 2018-10-02:
I deleted the folder that file was in and then opened atom again. The file
you refer to was pointing at python2.7 after that. It installed pymakr
after that. Thank you for your reply.

closed 2018-10-02 by RalphHogenbirk