pycom/pymakr-atom #100: Unable to install Pymakr Atom packages

adamticz, 2018-09-30:
this is the error what I receive:

 serialport@6.2.2 install C:\Users\I327221\AppData\Local\Temp\apm-install-dir-118830-16536-fnt0r1.7zr7\node_modules\pymakr\node_modules\serialport
> prebuild-install || node-gyp rebuild

C:\Users\I327221\AppData\Local\Temp\apm-install-dir-118830-16536-fnt0r1.7zr7\node_modules\pymakr\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
prebuild-install WARN install No prebuilt binaries found (target=2.0.7 runtime=electron arch=x64 platform=win32)
gyp info it worked if it ends with ok
gyp verb cli [ 'C:\\Users\\I327221\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\bin\\node.exe',
gyp verb cli   'C:\\Users\\I327221\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js',
gyp verb cli   'rebuild' ]
gyp info using node-gyp@3.4.0
gyp info using node@8.9.3 | win32 | x64
gyp verb command rebuild []
gyp verb command clean []
gyp verb clean removing "build" directory
gyp verb command configure []
gyp verb check python checking for Python executable "C:\Users\I327221\AppData\Roaming\Python\Python37\Scripts" in the PATH
gyp verb `which` failed Error: not found: C:\Users\I327221\AppData\Roaming\Python\Python37\Scripts
gyp verb `which` failed     at getNotFoundError (C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:80:29)
gyp verb `which` failed     at C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:89:16
gyp verb `which` failed     at C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\index.js:42:5
gyp verb `which` failed     at C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\windows.js:36:5
gyp verb `which` failed     at FSReqWrap.oncomplete (fs.js:152:21)
gyp verb `which` failed  C:\Users\I327221\AppData\Roaming\Python\Python37\Scripts { Error: not found: C:\Users\I327221\AppData\Roaming\Python\Python37\Scripts
gyp verb `which` failed     at getNotFoundError (C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:80:29)
gyp verb `which` failed     at C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:89:16
gyp verb `which` failed     at C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\index.js:42:5
gyp verb `which` failed     at C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\windows.js:36:5
gyp verb `which` failed     at FSReqWrap.oncomplete (fs.js:152:21) code: 'ENOENT' }
gyp verb could not find "C:\Users\I327221\AppData\Roaming\Python\Python37\Scripts". checking python launcher 
gyp verb could not find "C:\Users\I327221\AppData\Roaming\Python\Python37\Scripts". guessing location 
gyp verb ensuring that file exists: C:\Python27\python.exe
gyp ERR! configure error 
gyp ERR! stack Error: Can't find Python executable "C:\Users\I327221\AppData\Roaming\Python\Python37\Scripts", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\node-gyp\lib\configure.js:449:14)
gyp ERR! stack     at C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\node-gyp\lib\configure.js:404:11
gyp ERR! stack     at C:\Users\I327221\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\graceful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Windows_NT 10.0.16299
gyp ERR! command "C:\\Users\\I327221\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\I327221\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\I327221\AppData\Local\Temp\apm-install-dir-118830-16536-fnt0r1.7zr7\node_modules\pymakr\node_modules\serialport
gyp ERR! node -v v8.9.3
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\I327221\AppData\Local\Temp\apm-install-dir-118830-16536-fnt0r1.7zr7\package.json'
npm WARN apm-install-dir-118830-16536-fnt0r1.7zr7 No description
npm WARN apm-install-dir-118830-16536-fnt0r1.7zr7 No repository field.
npm WARN apm-install-dir-118830-16536-fnt0r1.7zr7 No README data
npm WARN apm-install-dir-118830-16536-fnt0r1.7zr7 No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\I327221\.atom\.apm\_logs\2018-09-30T12_33_02_653Z-debug.log

RalphHogenbirk, 2018-10-01:
Seems apm is using python 3 instead of 2.7. You can check this by running `apm --version` and looking at the python version.
To tell APM to use 2.7, use the following command:
`echo "python=/usr/bin/python2.7" >> ~/.atom/.apmrc`

Let me know if that helps

RalphHogenbirk, 2018-10-01:
This probably was another error that has just been solved in v1.4.4. Let me know if updating solves your issue as well.

closed 2018-11-01 by RalphHogenbirk