pycom/pymakr-atom #99: Installing/updating pymakr 1.4.3 failed; msbuild.exe to blame?

eymas, 2018-09-30:
It's been bugging me for a while, but I cannot update nor (re)install the latest version of the plugin. From what I can gather it seems to break upon calling msbuild.exe, although I haven't gotten much success in figuring out exactly why. Installing the previous version went without any hassle.
I'll make an attempt at reinstalling each component (python, atom et cetera.) in the meantime.

Here's the full log below:

`> serialport@6.2.2 install C:\Users\Tom\AppData\Local\Temp\apm-install-dir-118830-4792-tpq184.e6mdl\node_modules\pymakr\node_modules\serialport
> prebuild-install || node-gyp rebuild

C:\Users\Tom\AppData\Local\Temp\apm-install-dir-118830-4792-tpq184.e6mdl\node_modules\pymakr\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
MSBUILD : error MSB3428: Could not load the Visual C++ component "VCBuild.exe". To fix this, 1) install the .NET Framework 2.0 SDK, 2) install Microsoft Visual Studio 2005 or 3) add the location of the component to the system path if it is installed elsewhere.  [C:\Users\Tom\AppData\Local\Temp\apm-install-dir-118830-4792-tpq184.e6mdl\node_modules\pymakr\node_modules\serialport\build\binding.sln]

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
prebuild-install WARN install No prebuilt binaries found (target=2.0.7 runtime=electron arch=x64 platform=win32)
gyp info it worked if it ends with ok
gyp verb cli [ 'C:\\Users\\Tom\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\bin\\node.exe',
gyp verb cli   'C:\\Users\\Tom\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js',
gyp verb cli   'rebuild' ]
gyp info using node-gyp@3.4.0
gyp info using node@8.9.3 | win32 | x64
gyp verb command rebuild []
gyp verb command clean []
gyp verb clean removing "build" directory
gyp verb command configure []
gyp verb check python checking for Python executable "python2" in the PATH
gyp verb `which` failed Error: not found: python2
gyp verb `which` failed     at getNotFoundError (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:80:29)
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:89:16
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\index.js:42:5
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\windows.js:36:5
gyp verb `which` failed     at FSReqWrap.oncomplete (fs.js:152:21)
gyp verb `which` failed  python2 { Error: not found: python2
gyp verb `which` failed     at getNotFoundError (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:80:29)
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:89:16
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\index.js:42:5
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\windows.js:36:5
gyp verb `which` failed     at FSReqWrap.oncomplete (fs.js:152:21) code: 'ENOENT' }
gyp verb check python checking for Python executable "python" in the PATH
gyp verb `which` failed Error: not found: python
gyp verb `which` failed     at getNotFoundError (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:80:29)
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:89:16
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\index.js:42:5
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\windows.js:36:5
gyp verb `which` failed     at FSReqWrap.oncomplete (fs.js:152:21)
gyp verb `which` failed  python { Error: not found: python
gyp verb `which` failed     at getNotFoundError (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:80:29)
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\which\which.js:89:16
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\index.js:42:5
gyp verb `which` failed     at C:\Users\Tom\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\isexe\windows.js:36:5
gyp verb `which` failed     at FSReqWrap.oncomplete (fs.js:152:21) code: 'ENOENT' }
gyp verb could not find "python". checking python launcher 
gyp verb check python launcher python executable found: "C:\\Python27\\python.exe"
gyp verb check python version `C:\Python27\python.exe -c "import platform; print(platform.python_version());"` returned: "2.7.15\r\n"
gyp verb get node dir compiling against --target node version: 2.0.7
gyp verb command install [ '2.0.7' ]
gyp verb install input version string "2.0.7"
[... 236 lines cut ...]
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Users\\Tom\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Tom\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Tom\AppData\Local\Temp\apm-install-dir-118830-4792-tpq184.e6mdl\node_modules\pymakr\node_modules\serialport
gyp ERR! node -v v8.9.3
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\Tom\AppData\Local\Temp\apm-install-dir-118830-4792-tpq184.e6mdl\package.json'
npm WARN apm-install-dir-118830-4792-tpq184.e6mdl No description
npm WARN apm-install-dir-118830-4792-tpq184.e6mdl No repository field.
npm WARN apm-install-dir-118830-4792-tpq184.e6mdl No README data
npm WARN apm-install-dir-118830-4792-tpq184.e6mdl No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Tom\.atom\.apm\_logs\2018-09-30T12_02_21_200Z-debug.log`

stefanbrudny, 2018-09-30:
Same here, I must say I am really fed up with atom / pymakr / npm / ms never working.  Versions same, clean install. 

gyp verb "Release" dir needed to be created? null
gyp verb copying "iojs.lib" for x64 C:\Users\stefan\.atom\.node-gyp\.node-gyp\iojs-2.0.7\Release\iojs.lib
gyp info spawn C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
gyp ERR! build error 
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\stefan\AppData\Local\atom\app-1.31.1\resources\app\apm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Windows_NT 10.0.16299
gyp ERR! command "C:\\Users\\stefan\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\stefan\\AppData\\Local\\atom\\app-1.31.1\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\stefan\AppData\Local\Temp\apm-install-dir-118830-28632-uhaav8.beuf\node_modules\pymakr\node_modules\serialport
gyp ERR! node -v v8.9.3
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok

RalphHogenbirk, 2018-10-01:
Hi. It seems like a few others are struggling with this on 1.4.3. I've spotted the mistake, which is that the serialport dependancy shouldn't even be in the package.json since it fails on most windows machines (as seen in your stacktraces). On windows it always falls back to the included precompiled version of serialport.

I've just released 1.4.4 with this small fix. Please let me know if this solves your issues.

eymas, 2018-10-01:
> Hi. It seems like a few others are struggling with this on 1.4.3. I've spotted the mistake, which is that the serialport dependancy shouldn't even be in the package.json since it fails on most [...]

Yes it has, v1.4.4 installed successfully through apm and I've tested it by making & uploading some testing code to a LoPy4.
Glad to hear the mistake was spotted quickly and thanks for the response!

closed 2018-10-01 by eymas