pycom/pymakr-atom #101: Help - Pymakr 1.4.3 Update or Installation Fails on Atom

Stevo52, 2018-10-01:
I have been using Atom with Pymaker 1.4.2 quite happily for a few weeks without issues.

The was an advice in Atom that the Pymakr package had an update available. I tried to do the update but it failed, several times.

I have tried a number of things and have eventually completely removed the Pymakr 1.4.2 package and Atom itself, then started again with a fresh Atom install and then the Pymakr 1.4. package installation. It failed again (several times).

So (as I am new to Github) what do I do now? Do I need to upload the install log ->
[2018-10-01T01_54_37_875Z-debug.log](https://github.com/pycom/pymakr-atom/files/2432348/2018-10-01T01_54_37_875Z-debug.log)

Can I roll back to Pymakr 1.4.2 so I can atleast continue with developing my code? Where do I find that?

Thanks in advance for any assistance.

Stevo52

RalphHogenbirk, 2018-10-01:
Hi @Stevo52, I have a few more reports about 1.4.3, we'll try to figure that out today. For now you could roll back using the command `apm install pymakr@1.4.2`

Stevo52, 2018-10-01:
Thanks for the response, I'll have to figure out how and where to run that apm string from..

RalphHogenbirk, 2018-10-01:
You can run it from your terminal / command line.

Stevo52, 2018-10-01:
Just figured it out..had to add the path :-)

Stevo52, 2018-10-01:
Thanks for the help, I have pymakr@1.4.2 running again and can carry on with my project..

Hope you resolve the issue with the update version..

RalphHogenbirk, 2018-10-01:
1.4.4 has just been released, which should solve your issue.

Stevo52, 2018-10-01:
1.4.2 ran a test routine all night (I'm in Australia). No problems. I tried to the update this morning but although there were no errors or failure reported, it doesn't seem to have actually updated?

Stevo52, 2018-10-01:
Restarted Atom and tried again..
[image]

Stevo52, 2018-10-01:
A complete system reboot, restart Atom and tried the update again and it seems to be successful. Won't be able to try it until a bit later, when I am in the workshop.

Stevo52, 2018-10-02:
Hi Ralph, I have successfully updated to pymakr@1.4.4 on all three PC's I use with Atom for development. What I have noted though is that the update doesn't appear to be successful initially (took 2 attempts on 2 machines but only 1 for the other) and no version detail is displayed until Atom is restarted and the installed packages are checked. But now installed and running successfully on the two machines in my workshop with Pycom Expansion 3 boards and Gpy's at present. Thanks for sorting the issue out so quickly.

RalphHogenbirk, 2018-10-02:
Good to hear @Stevo52. 

I'll keep an eye out for people having this same issue where they have to attempt the install 2 times. I know atom is kind of harsh with caching sometimes, and Windows sometimes blocks write access to the serialport-win for nodejs. Not sure how to recreate or solve that issue, but for now rebooting atom like you did, or even manually clearing the pymakr extension folder and cache folders, solves it.

closed 2018-10-02 by RalphHogenbirk