pycom/pymakr-atom #102: Cannot tape any command line on REPL console

bricedenicedu69, 2018-10-01:
Hello! Since a few days now, I can't type any command line on the Pymark, although my Pycom board is connected. I even have the ''Connected'' sign so I guess that is not the problem. Then, when I try to run the code, it's written ''Running mycode...'' but for a really long long time and even after that, nothing happens and the code is still running. I first wanted to type my command lines on the consol separately but it's not working on worrying me.
I don't know if I made myself clear enough but I can provide you with further information if needed.
Thanks

RalphHogenbirk, 2018-10-01:
Hi @bricedenicedu69, which OS are you using, and which Pymakr/Atom versions? 

Usually this happens when there is code running on the board that's blocking access to the REPL. You could try ctrl-c to stop the code after connecting, or even ctrl-f to do a safe boot.
Another common cause of this issue is when there are jumpers missing from the expansion board. Have at least the top 4 jumpers (tx, rts, rx, cts) placed to have a all terminal features.

bricedenicedu69, 2018-10-02:
Hi Ralph and thanks for answering so fast.
I am using Atom 1.31.1 and Pymakr 1.4.4 which are the most recent versions I think. I did a safe-boot by connecting G20 (P12) to 3V3 and pressing the reset button. The LED started blinking orange faster and faster. When I checked the consol, I had multiple messages : load:0x3ffff8020,... the last one being : E(8006) esp_image:Checksum failed. Calculated 0x24 read 0x72. I guess this did not work.
I shall try the other solutions you proposed but I do have the 7 jumpers (tx,rts,rx,cts) included, even if I did not cable them nor use them really.
With best regards

RalphHogenbirk, 2018-10-02:
The 'checksum failed' message makes me think the firmware on the device is corrupt. You could try to flash the firmware again using the [firmware updater tool](https://docs.pycom.io/gettingstarted/installation/firmwaretool). Let me know if that works.

bricedenicedu69, 2018-10-04:
I followed your advice and it worked! Thanks for your help.
With best regards,
Brice

closed 2018-10-04 by RalphHogenbirk