pycom/pymakr-atom #105: pymakr window too small

Kibabalu, 2018-11-02:
pymakr window too small. It's not possible to increase size. The attached picture shows the biggest size possible.
[image]

sdaniel55, 2018-11-02:
Hi @Kibabalu 👋 

You can resize the panel by drag the top of it up (see below)

[image]

emmanuel-florent, 2018-11-24:
Same thing here could not resize PyMakr window, PyMakr window too small, remove atom-ide-ui package and restart make it works like it should.

RalphHogenbirk, 2018-11-27:
@emmanuel-florent As seen above, it's a known issue. I've created an issue in the atom-ide-ui github to ask for help on it, because I can't find the exact cause of it.

lewisoaten, 2019-04-27:
Also having this issue. I found the other issue that was made on atom-ide-ui (https://github.com/facebookarchive/atom-ide-ui/issues/323) but that repo is now archived. Has this actually been fixed. I can work around by disabling atom-ide-ui.