pycom/pymakr-atom #107: 1.4.7 Installation Failed

SandrisIntelZone, 2018-12-13:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Install latest stable Atom 1.33.0 x64 
2. Install Package Pymakr 1.4.7
3.

## What happens?
Failed to install 

## What were you expecting to happen?

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
https://gist.github.com/SandrisIntelZone/360e8fe68c5e434a945e3e0bfd26ddf1

## Any other comments?

## What versions of software are you using?

**Operating system:**
Windows  10

**Atom version:**
 1.33.0 x64 

**Pymakr version:**
 1.4.7

fusspawn, 2018-12-13:
This looks to be atom serving up SSL connections that the client cant handle. handshake is failing due to the server wanting an SSL version the client cant deal with. 

Either we are out of date. or Atom is.

RalphHogenbirk, 2018-12-17:
I tried installation using the same versions on a x32 machine and that worked well. Will try it on a x64 later. In the meantime, try closing atom and deleting the pymakr package folder (~/.atom/packages/pymakr) completely, before trying to install the plugin again. Sometimes there are permission issues on some of the subfolders.

SandrisIntelZone, 2018-12-17:
I solved it by installing pymakr manually. The problem was that on my PC  is a corporate proxy that is a cause of this problems.

closed 2018-12-17 by SandrisIntelZone