pycom/pymakr-atom #108: Write errors when looping via micropython.schedule() on ESP32

psy0rz, 2018-12-20:
## What are the steps to reproduce this issue?
1. Create a function that schedules itself via micropython.schedule(loop,None)
2. Try to upload files

## What happens?

Write errors and retries. Sometimes it cant read the status file and tries to reupload all files.

## What were you expecting to happen?

Uploading the files and rebooting.

## Any other comments?

I also tried just using a while True: loop, but then it also gets timeouts and cant do anything?

## What versions of software are you using?

esp32-20180511-v1.9.4.bin

**Operating system:**
 Ubuntu 18.04.1 LTS

**Atom version:**
Atom    : 1.32.2
Electron: 2.0.9
Chrome  : 61.0.3163.100
Node    : 8.9.3

**Pymakr version:**

1.4.7

RalphHogenbirk, 2018-12-20:
Are you running a Pycom board or another ESP32 board? When using the first one, make sure you have the 'safe-boot before upload' setting enabled in Pymakr. This makes sure that the board doesn't run any code while uploading, which could interfere with the process (and result in timeouts/failures and retries). If it's another board, try safe booting manually if supported by the board.

Let me know if that solves your issue.

psy0rz, 2018-12-20:
Nope another ESP32 board (wemos d32).

Safe-boot indeed doesnt work. If I have to do a manual action, that kind of defeats the purpose. I could also just stop my loop to make it work.

I was hoping for an automated solution. Is there any way to make pymakr execute custom code? Can i hack it in somewhere? That way i can stop the loop.

Or does it already call some kind of stub befrore uploading, that i can override?

RalphHogenbirk, 2018-12-22:
@psy0rz does ctrl-c not stop your code? Pymakr should do a ctrl-c before uploading to stop the code. Or do you need to call some specific function to stop it properly? 

If the latter is the case, you can hack it in quite easily. Quick and dirty solution to test if it works, is to customise the 'stop_running_programs_double()' function in the file `~/.atom/packages/pymakr/lib/board/pyboard.js`. This function is executed before starting the upload procedure (`~/.atom/packages/pymakr/lib/board/sync.js:343`) with a timeout of 500ms. 

I'd change the code in this function to something like this, sending your own code instead of the double CTRL_C command:
```
var command =
        "import whatever\r\n" +
        "mycode.stop()\r\n"
        
    this.send_wait_for(command,">>>",function(err){
      if(cb) cb(err)
    },timeout)
```
Possibly increase the timeout a bit, if your code takes more than 500ms to stop.

closed 2020-02-18 by pedroknup