pycom/pymakr-atom #109: Pymakr incompatible with Atom (v 1.34.0)

krisavi, 2019-01-09:
Seems like yesterday was new version of atom released, of course it makes the Pymakr not to work with latest Atom again.

## What versions of software are you using?
**Operating system:**
Windows 10
**Atom version:**
1.34.0
**Pymakr version:**
1.4.7

RalphHogenbirk, 2019-01-09:
Thanks for noticing, I'll test and update Pymakr with 1.34 compatibility first thing tomorrow morning.

RalphHogenbirk, 2019-01-10:
@krisavi Atom 1.4.8 has just been released, including compatibility with 1.34. Let me know if it works well for you.

krisavi, 2019-01-10:
I was able to manually install and build it. Seems to run fine, well at least no errors popping up. Might be as easy as bumping up the version compatibility.

RalphHogenbirk, 2019-01-10:
@krisavi yes, that's exactly what I did :) There are no code changes to fix the compatibility, I just bumped the versio number (and included one other bugfix that was unrelated)

krisavi, 2019-01-10:
@RalphHogenbirk  Version 1.4.8 seems to work fine for me as well.
Thank you! 👍 :)

closed 2019-01-28 by RalphHogenbirk