pycom/pymakr-atom #111: Uncaught TypeError: Cannot read property 'writeln' of undefined

kiik, 2019-01-28:
1. Open atom with pymakr 1.4.9 installed.
2. Attempt to open terminal window, results in exception.

**Atom**: 1.34.0 x64
**Electron**: 2.0.16
**OS**: Ubuntu 18.04.1
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.9

### Stack Trace

Uncaught TypeError: Cannot read property 'writeln' of undefined

```
At /home/globalreader/.atom/packages/pymakr/lib/pymakr.js:474

TypeError: Cannot read property 'writeln' of undefined
    at /packages/pymakr/lib/pymakr.js:474:25
    at /packages/pymakr/lib/board/pyboard.js:65:14
    at /packages/pymakr/lib/main/settings-wrapper.js:141:16
    at /packages/pymakr/lib/connections/pyserial.js:144:11
    at FSReqWrap.oncomplete (fs.js:135:15)
```

### Commands

```
  2x -0:03.9.0 core:close (atom-pane.pane.active)
```

### Non-Core Packages

```
atom-beautify 0.33.4 
busy-signal 2.0.0 
intentions 1.1.5 
language-elixir 0.22.1 
language-slim 1.0.0 
linter 2.3.0 
linter-cppcheck 0.2.3 
linter-ui-default 1.7.1 
project-manager 3.3.6 
pymakr 1.4.9 
symbols-tree-view 0.14.0 
```

RalphHogenbirk, 2019-01-28:
HI @kiik, thanks for reporting this. I checked the code and there is indeed a bug on that line (it references 'this' instead of '_this'). The line is supposed to print the following line in the terminal: "Address not configured. Please go to the settings to configure a valid address or comport". So If you configure the address or enable Auto Connect from the package settings, your error should disappear.
I've fixed the bug and will release a hotfix for it.

RalphHogenbirk, 2019-01-28:
Has been fixed in release v1.4.10.

closed 2019-01-28 by RalphHogenbirk