pycom/pymakr-atom #112: Uncaught TypeError: path must be a string or Buffer

jirikrepl, 2019-01-29:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.34.0 x64
**Electron**: 2.0.16
**OS**: Mac OS X 10.14.2
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.10

### Stack Trace

Uncaught TypeError: path must be a string or Buffer

```
At fs.js:904

TypeError: path must be a string or Buffer
    at Object.fs.readdirSync (fs.js:904:18)
    at Object.fs.readdirSync (ELECTRON_ASAR.js:608:28)
    at ProjectStatus.__get_local_files (/packages/pymakr/lib/board/project-status.js:88:15)
    at ProjectStatus.__get_local_files_hashed (/packages/pymakr/lib/board/project-status.js:93:20)
    at /packages/pymakr/lib/board/project-status.js:18:35)
    at /packages/pymakr/lib/board/sync.js:37:27)
    at Pymakr.sync (/packages/pymakr/lib/pymakr.js:643:22)
    at Pymakr.upload (/packages/pymakr/lib/pymakr.js:612:10)
    at HTMLElement.pymakrUpload (/packages/pymakr/lib/main.js:28:45)
    at CommandRegistry.handleCommandEvent (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:350017)
    at KeymapManager.t.exports.KeymapManager.dispatchCommandEvent (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:1212945)
    at KeymapManager.t.exports.KeymapManager.handleKeyboardEvent (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:1209079)
    at WindowEventHandler.handleDocumentKeyEvent (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:285196)
```

### Commands

```
     -0:23.7.0 core:paste (input.hidden-input)
     -0:20.3.0 pymakr:upload (textarea.xterm-helper-textarea)
```

### Non-Core Packages

```
goto-definition 1.3.4 
hyperclick 0.1.5 
pymakr 1.4.10 
```

jirikrepl, 2019-01-29:
trying to upload a file without project will do this error

[image]

RalphHogenbirk, 2019-01-29:
Thanks for reporting @jirikrepl, I've found the issue and pushed a fix on the develop branch (commit 19f48b6). I'll make sure it's included in the next release.

closed 2019-01-29 by RalphHogenbirk