pycom/pymakr-atom #116: Pymakr not connecting

AlexSartori, 2019-04-04:
My Pymakr console is stuck on `Connecting to /dev/ttyUSB0...`

The device address is correct and my user is in the `dialout` group, in fact `screen /dev/ttyUSB0 115200` brings up the REPL. No jumper wires are connected to the board.

## What are the steps to reproduce this issue?
1. Connect the LoPy Expansion Board 2.0 with a USB cable
2. Launch Atom

## What happens?
The console window is stuck on `Connecting to /dev/ttyUSB0...` and the connection indicator won't change from "Disconnected [red X]". Retrying to connect leads to these messages:
```
Connecting to /dev/ttyUSB0...
Connection attempt canceled
Connecting to /dev/ttyUSB0...
> Failed to connect (Error: Port is not open). Click here to try again.
> Failed to connect (Error: Port is not open). Click here to try again.
```

## What were you expecting to happen?
Have a REPL appear in the console window and the connection indicator to turn into "Connected [green V]".

## What versions of software are you using?

**Operating system:** Fedora 28, 4.20.8-100.fc28.x86_64

**Atom version:** 1.28.1

**Pymakr version:** 1.4.12

RalphHogenbirk, 2019-04-04:
Hi @AlexSartori, first thing i'd try is updating Atom to the latest version (1.35.1). 1.28 is a bit outdate, it might be causing issues.

let me know if that helps

AlexSartori, 2019-04-04:
Definitely solved the issue.

Don't know how I thought I had the last version, thanks a lot.

closed 2019-04-04 by AlexSartori

RalphHogenbirk, 2019-04-04:
No worries, glad it's solved!

swathipai13, 2020-06-30:
Hi facing a similar issue, and this hasn't solved. I am using Visual Studio Code instead of Atom.
My Pymakr console is **stuck on Connecting to COM14.**
the steps to reproduce this issue?
>Connect an esp8266 board via USB
Made required changes in pymakr, project and global settings for detection of COM14.
Specs: VSCode( version up to date, checked again)
Pymakr(v1.1.6)
Windows 10