pycom/pymakr-atom #122: Pycom Pymakr Atom no project and can't connect wipy

Stayfire, 2019-06-05:
Hi, I'm using atom with a pycom for a project. 

Steps i'm doing
1. Launch Atom
2. Plug the usb on pysense
3. Connect 

I'm stuck here, a message says: Connecting to COM8...
I wait several seconds and an error message appears: Connection error: Error: timeout
Yet atom tells me that the card is connected.

In addition, I even have loaded a project, atom does not detect it.

[image]

Another thing, when I click on Run and then upload, the upload and download buttons disappear. 
The only way to make them reappear is to restart atom since the ctrl + c does not work

[image]

I would like to know why the card does not want to connect (I configured the card in COM8 following the tutorial), to know if it is the version on which I am that does not work or if the problem comes from elsewhere and find out why I have an error message when I want to upload: 

[image]

I use the following versions:
Atom : 1.37.0 x64
Pymakr: 1.4.13                    I tried with version 1.0.1 but it's still the same
Pycom, Wipy, and Pysense
Python 3.7
Windows 7 pro

Thank you for your help !

RalphHogenbirk, 2019-06-05:
Hi @Stayfire,

The disappearing buttons is a known issue, will be fixed soon!

About the connection issue: Could you try connecting using Putty or Pymakr for VSCode? That way we can exclude issues with the board.
Also, which firmware version are you using on the board? Make sure you are on the latest version on both the Wipy as well as the Pysense board.

Stayfire, 2019-06-05:
@RalphHogenbirk, Thank for your answer. 

I use the firmware given in the online help, the version is 1.15.1

I managed to connect the card and upload a program that is:

[image]

It allows to flash the led in different colors; The problem is that the program starts correctly but only once, then I have to disconnect the card and reconnect it in usb. In addition the program is supposed to be an infinite loop (with the while True) but when the first cycle ends, the led goes out but does not come back and I can not write / type any more commands in the REPL of Pymakr. (The >>> do not appear anymore, I have to disconnect the card, unplug it, reconnect it and click on connect).

I will try on VS Code, thank you for your help

Stayfire, 2019-06-05:
So I just tried on VSCode, the program runs but it stops at the second cycle. In addition I still have the same problem the card "disconnects" so when I run the program I can not stop it or run twice in a row.

[image]

As you can see on the screen above I launched VS Code and the connection was not established (Failed to connect), I unplugged and reconnected the card in usb, the >>> appear but I still gets an error. and when i unplugged and plugged in again it worked but i got an error again when i ran the program.

Maybe I have misconfigured ports but I followed the guide step by step. I do not know at all where I was wrong and why the software marks me Failed to connected (Error: Port is not open). By the way I use a Wipy 3.0 and a Pysense V1.0. This is the first time I use a pycom card !

[image]

Thank you for your attention !

Stayfire, 2019-06-07:
I solved it, I wasn't using the last version of the firmware. Thank you for your help.

closed 2019-06-07 by Stayfire