pycom/pymakr-atom #124: Pymakr 1.4.13 incompatible with Atom 1.38.2

TomBurg, 2019-06-19:
I was updating pymakr plug-in to version 1.4.13 but then was told it was incompatible.
Is the problem that I updated Atom to 1.38.2 x64 ?
Do I uninstall Atom 1.38 and replace with 1.37 or is there a fix for pymakr ? (Version 1.4.14 ??)

**Operating system:** Windows 10 Home Version 1809

**Atom version:** 1.38.2

**Pymakr version:** 1.4.13

RalphHogenbirk, 2019-06-19:
Latest version of Pymakr is 1.4.14, which was released yesterday and is compatible with 1.38 and and 1.39-beta.

closed 2019-06-19 by RalphHogenbirk

TomBurg, 2019-06-19:
> Closed #124 <https://github.com/pycom/pymakr-atom/issues/124> .
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> https://www.avast.com/antivirus

Thank you, Thank You, THANK YOU!!