pycom/pymakr-atom #125: Unable to connect to USB device from within Atom using esp32-WROOM-32

jdcrunchman, 2019-07-03:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1.
2.
3.

## What happens?

## What were you expecting to happen?

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## What versions of software are you using?

**Operating system:**

**Atom version:**

**Pymakr version:**

jdcrunchman, 2019-07-03:
I'm connecting my Mac to this esp32 board,   then after installing PyMakr,   and specifying the USB port address of:   /dev/tty.SLAB_USBtoUART

Then, when I try to connect to it,   I get:    AutoConnect no device available.
I know I'm not using the PyCom board,  but using the ESP32-WROOM-32 board,   is there
a way I can get PyMakr to work with this Atom plugin.

Please reply to jdcrunchman@gmail.com

John

RalphHogenbirk, 2019-07-12:
@jdcrunchman disable auto_connect in the global settings. The autoconnect feature ignores any manual serialport addresses that you configure using the 'address' config field.

closed 2019-07-12 by RalphHogenbirk

jdcrunchman, 2019-07-12:
I just got this reply,    you didn't specify which app that has the auto-connect.  Were you referring to the Atom IDE?

jdcrunchman, 2019-07-12:
Ok,  it was in Atom IDE.   I found the setting and fixed it,  so I can now use Atom,   YAY!  I'm so happy.  I never would have figured that out.    Thanx a heap.

John