pycom/pymakr-atom #127: Failed to load the pymakr package

Jakub0721, 2019-07-24:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.39.0 x64
**Electron**: 3.1.10
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.14

### Stack Trace

Failed to load the pymakr package

```
At The module '\\?\C:\Users\dabko\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Error: The module '\\?\C:\Users\dabko\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:146719)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:3:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:18:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.0/resources/app/static/<embedded>:11:147410)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.4.14 
```

How to solve this problem?

Dielectric2015, 2019-07-24:
I'm getting the same error, from Atom and from VSCode.

JDMaughan, 2019-07-24:
atom needs to be told to use 2.7 instead of 3.x.Common issues

Failed to load package: Cannot find module 'serialport'
In some cases this is caused by the Atom Package Manager (apm) using Python 3.x, while node-gyp (used for compiling the serialport lib) needs Python 2.x. To confirm this, apm --version can be run to check which Python version apm is using.

Solution: Tell the package manager to use python 2 instead. Running the following command switches apm to 2.7:

echo "python=/usr/bin/python2.7" >> ~/.atom/.apmrc

Now reinstall Pymakr or run apm install from the Pymakr package located in ~/.atom/packages/pymakr

Dielectric2015, 2019-07-24:
I don't think that's the issue, as it just did that today.  I've had the plugin running fine for months, then today I logged out, logged back in, and got the error on the next launch.  I haven't had Python 2.7 installed for a really long time, so just for fun I installed it, changed the .apmrc as you showed (with edits for Windows paths), deleted Pymakr  from the Atom directory and get the same error trying to re-install from Atom.

MatiasDurbano, 2019-07-25:
idem, im getting the same error an hour ago

JDMaughan, 2019-07-25:
Thanks for the response.  This was my first time installing pymakr in atom
and I just reviewed the common issues section.

Will someone update the package anytime soon?

Dessha, 2019-07-25:
I have the same problem. seems to be caused by atom auto updating. which is fun when i set it to only update manually....

it first happened to my work pc and after switching to my laptop it happened again.

CeruleanMars, 2019-07-25:
The same error appeared for me. I uninstalled Atom and reinstalled an older release (1.38.2), which now appears to be working with Pymakr.

ClarkP57, 2019-07-25:
> The same error appeared for me. I uninstalled Atom and reinstalled an older release (1.38.2), which now appears to be working with Pymakr.

This worked for me as well.

aliubi, 2019-07-25:
Apparently Atom updates whether you want it to or not.  It looks like serialport.node needs to be recompiled.  The Python version modification has no effect on my machine.

aesilab, 2019-07-26:
Someone from Pycom can say something about this issue??? It seems it is a misunderstanding between pymakr and the new release of Atom.
-------------------------------------------------------------------------

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.39.1 x64
**Electron**: 3.1.10
**OS**: Unknown Windows version
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.14

### Stack Trace

Failed to load the pymakr package

```
At The module '\\?\C:\Users\Giorgio Giaccaglini\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Error: The module '\\?\C:\Users\Giorgio Giaccaglini\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:146719)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:147410)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:150958)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:71:46)
    at require (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:146719)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:3:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:18:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.39.1/resources/app/static/<embedded>:11:147410)
```

### Commands

```
```

### Non-Core Packages

```
compare-files 0.8.1 
pymakr 1.4.14 
script 3.18.1 
```

HenroRitchie, 2019-07-26:
Same here - worked for months without a problem, and now starting Atom with this issue. Not helping as I am in die middle of a project!!

aliubi, 2019-07-26:
Weigh in here as well, it seems everyone at PyCom is on vacation this week

[https://forum.pycom.io/topic/5040/is-rollback-possible](https://forum.pycom.io/topic/5040/is-rollback-possible)

RalphHogenbirk, 2019-07-27:
@aliubi is right, I'm typing this from a tent in Norway. Sorry about this. I guess atom chose to up the electron version in the wrong week :) it's a relatively easy fix, but I have to get a chance to charge my laptop somewhere, run the fix and push the update. Will try to do that asap. And after the holiday, get someone else in the pycom office the credentials to push new versions when I'm not around. 

By the way, alternative is VSCode. or if you need atom, simply downgrade by downloading it from their git releases page. Should stay that way if you disable auto update.

groger57, 2019-07-27:
Atom 1.38.2 (and up to latest) crashes (both 64 and 32) after pymakr is installed. Ouch. Counter-productive. Any testing done before release? Will have to go with VS code.

coolhouse, 2019-07-30:
Any word on the fix for this issue?

groger57, 2019-07-30:
I just stopped using Atom and pymakr. Cannot afford to waste anymore time on silliness. Now using ESP32 Expressif board, with uPycraft. Seems to be working just fine!

MJtuks, 2019-07-31:
Same issue here. Any indication of when this will be fixed?

HenroRitchie, 2019-07-31:
@groger57 I am also concerned about this dev environment. I am always scared of updating pymakr, or Atom as updates usually break the dev environment. Unfortunately, Atom was set to auto-update this time around.

jdcrunchman, 2019-08-01:
I asked this question about 5 days ago,   have not gotten any response or help from this yet.
I have a serious vision problem.   The font size of the ESP32 console is far too small for my 76 year old eyes.    It would be so much better for me,  if I could read the console without squinting my eyes,  or using a magnifying class.

Is there anyway I can change the font size.    The PyMar plugin has a font setting,   but when I try and set it,   nothing happens!   Is this a bug,  or in my insane stupidity, have I overlooked a "SAVE" button to save the font size?    Once I put in the larger font size in the PyMakr settings,   does it automatically update the font with no other action?

groger57, 2019-08-01:
you might try using a different resolution on Windows itself. I know this does not directly answer your question, but it might suffice until you get an answer.

Stevo52, 2019-08-03:
I have just come back to tmy projects after a period of illness and of course I can't get Pymakr to open. What is the status with this issue? Is it being fixed or is a roll back to Atom 1.13.82 or earlier, the only fix? Has Pycom not taken any action to repair this? Not a great advert for them if that is the case.

jdcrunchman, 2019-08-06:
not using WinBlows - I'm running a Mac.

John

MJtuks, 2019-08-07:
Anyone's Atom also auto-updating even with the setting turned off?

Stevo52, 2019-08-07:
Think you sent that to the wrong person. I didn't make that comment.

Kind Regards
Steve Williams
E: stevo7452@gmail.com
M: +61 (0) 410452855

JeanPoll, 2019-08-08:
Same issue here

aesilab, 2019-08-08:
issue opened 15 days ago... no answers... no solutions. Bad support from pycom.

HenroRitchie, 2019-08-08:
I suspect the Pycom team is on summer break/holiday. I can confirm rolling Atom back to 1.38.2 with pymakr 1.4.14 works.

Dielectric2015, 2019-08-09:
Has anyone tried the rebuild advice given in the error message?  I'm not hip enough to have used node in the past, but I tried fooling around in the plugin tree to no avail.  

My VSCode on Linux was broken on the same module but following the steps in Issue 53 fixed it:
https://github.com/pycom/pymakr-vsc/issues/53
It basically pulls in a new precompiled module, Atom apparently works differently.

Josverl, 2019-08-19:
Let be start by declaring my ignorance: I've never used Atom.

However you seem to be hitting issues cause by Atom moving to a newer version of Electron that uses a new ABI ( Application Binary Interface) , which has the nasty side effect of breaking native modules , i.e. the Serialport.
The temp fix was a 1:1 fix for the platform, i've since then built and tested a more extensive solution (for VSCode) 
I think this is now solved ( after acceptance my PyCom) for **Pymakr on VSCode**.  

But as both use Electron, I would say that the that the same can be applied to Atom as well by pre-packaging the correct files. ( and possible add some autodetectio of Atom / electron future versions) 
Its even possible to do this manually to you current version(s) of Atom by removing one file , and copying folder with the different native bindings for the different platforms.

The below may or may not work . As I said I have absolutely no experience at all with Atom. 
I have no clue on where Atom stores it extensions or node_modules 

## Here be dragons
do not attempt the below unless you are sure that you can do the below 
and if you can and want to take the risk thay you may need to reinstall Atom and any/all extentions and configuration.

I can provide you no support.
**If you are a paying customer of Pycom, please call support before you try this.**

1 `del <path to pymakr extension>/node_modules/@serialport/bindings/bindings.node_modules/bindings.node` 
Note: the path may/will differ somwhat from linux distro's / windows 
2. download the [serialport bindings I have packaged](https://github.com/Josverl/pymakr-vsc/releases/tag/MultiPlatform_vscode )
3. extract these into the native_modules folder somewhere  ( desktop / pendrive / home )
4. copy all files and folders in the native_modules folder into the node_modules folder.   
The result should NOT have a the native_modules folder itself inside the node_modules folder) 
5. Restart Atom 

Please let others know if this worked or not.

RalphHogenbirk, 2019-08-28:
Sorry for the extreem delay people. It's been holiday and business trip season for me. But in short, i've just released Pymakr 1.4.15, which is compatible with atom 1.40 and 1.41-beta. Should work for everyone, at the very least on windows and osx. 

@Josverl in his last comment is right, but atom works a bit differently. I've done something similar for atom, just using different versions to compile the bindings.node files to, and a different location to store them. Atom automatically copies the correct precompiled bindings.node file to the @serialport/bindings folder, if your local compile failed on install.

mgarrett1955, 2019-08-29:
Im having the same issue, just downloaded latest version from scratch. Have MAC OS Mohave, seems to have broken after OS update. I tried deleting .atom and restarting, then loading up the pymakr package.. eed to update a project, is there any updates on this issue?

RalphHogenbirk, 2019-09-02:
Hi @mgarrett1955, on the latest versions (atom 1.40 and pymakr 1.4.15) this issue should be solved. Could you check if you have those versions? If so and it's not working, can you send the error or stack trace that you are getting?

mgarrett1955, 2019-09-02:
I do have those versions, here is a stack trace:

Error: Could not locate the bindings file. Tried:
 → /Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/build/bindings.node
 → /Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Release/bindings.node
 → /Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/Debug/bindings.node
 → /Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Release/bindings.node
 → /Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/Release/bindings.node
 → /Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/build/default/bindings.node
 → /Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/compiled/10.2.0/darwin/x64/bindings.node
    at bindings (/Users/michael/.atom/packages/pymakr/node_modules/bindings/bindings.js:93:9)
    at Object.<anonymous> (/Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/darwin.js:1:191)
    at Object.<anonymous> (/Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/darwin.js:104:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:72:46)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146745)
    at Object.<anonymous> (/Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/index.js:10:22)
    at Object.<anonymous> (/Users/michael/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/index.js:17:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:72:46)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146745)
    at Object.<anonymous> (/Users/michael/.atom/packages/pymakr/node_modules/serialport/lib/index.js:2:17)
    at Object.<anonymous> (/Users/michael/.atom/packages/pymakr/node_modules/serialport/lib/index.js:17:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:72:46)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146745)
    at Object.<anonymous> (/Users/michael/.atom/packages/pymakr/lib/connections/pyserial.js:5:18)
    at Object.<anonymous> (/Users/michael/.atom/packages/pymakr/lib/connections/pyserial.js:7:30)

RalphHogenbirk, 2019-09-02:
Thanks for that @mgarrett1955 . The stacktrace reports it looked for bindings in: `/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Release/bindings.node` which is actually the correct folder. Can you check if it's indeed missing? A quick fix is to copy it there from `.atom/packages/pymakr/precompiles/serialport-osx`.

To debug, I'd be interested to see what the debug console outputs (open using cmd-opt-i) It should report something about repairing the serialport bindings, perhaps something is failing in that process.

closed 2020-02-18 by pedroknup