pycom/pymakr-atom #129: Failed to activate the pymakr package in windows 10

choc70, 2019-07-31:
was compiled against a different Node.js version using

NODE_MODULE_VERSION 57. This version of Node.js requires

NODE_MODULE_VERSION 64. Please try re-compiling or re-installing

the module (for instance, using npm rebuild or npm install).

jaketeater, 2019-07-31:
seems to be a duplicate of #127 

For me, the work around was to downgrade Atom to 1.38.2

Oct-opus, 2019-08-04:
Same here, 
Downgrading to 1.38.2 does fix it but you have to disable Atom auto update.
It's not rely a fix, just a temporary solution ...

sophiekovalevsky, 2019-08-06:
I can confirm that I had this bug. Before compiling again the package, I tried to uninstall pymark, and relaunch atom, it works. 

Atom version: 1.39.1

closed 2020-02-18 by pedroknup