pycom/pymakr-atom #131: Pymakr 1.4.14 Issue with atom 1.39.1

jo-taye, 2019-08-08:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

1. I was using atom 1.38.2 with pymakr 1.4.14 and everything was working fine. But atom automatically upgraded to the latest release 1.39.1 and right after doing so it produced the following error 
`\C:\Users\user\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).`

I found that reverting my atom installation to version 1.38.2 fixes the problem.

innoagg, 2019-08-14:
Atom just went to 1.40.0. Can we get an update so we can run pymakr? I can't figure out how to stop Atom from auto-updating.

teuler, 2019-08-16:
Same problem here. The next time Atom restarts, it did an autoupdate :-(

johnmcdnz, 2019-08-17:
I just had this and seem to have fixed it.  After the update and realising that Pymakr no longer worked.
1/  uninstall Atom using standard Windows app procedure.
2/  Go to here and download the old Atom setup
https://github.com/atom/atom/releases/tag/v1.38.2
3/  Allow to install.
4/  Make sure that when Atom starts, go to the About menu and DISABLE 'Automatically download updates.
5/  If it does download an update, delete the download from the directory
...AppData/Local/Atom
6/  Reinstall Pymakr 

have fun.  Hope this helps to fix.

RalphHogenbirk, 2019-08-28:
Sorry for the inconvenience. Atom 1.4.15 has just been released, which solves this problem.

jo-taye, 2019-08-29:
That's great news :)

teuler, 2019-08-30:
Yes, thanks!

closed 2019-11-27 by pedroknup