pycom/pymakr-atom #132: No pymakr version for latest Atom 1.40.1

samkalnins, 2019-08-24:
It's not possible to install pymakr in the latest production release of Atom (1.40.1) -- since pymakr specifies it only supports Atom <1.40.0

Can we have a pymakr version which supports >1.40.0?

Thanks!

[image]

RalphHogenbirk, 2019-08-28:
This has been solved with the release of 1.4.15 that was done this morning.

jdcrunchman, 2019-09-20:
Ralph...  this fix does not work.   I installed a new Atom,  1.40.1 X64.   Tried to install the new pymakr 1.4.16.    I get the usual error message....
[image]
as seen from this screenshot.   Has there, been any workarounds posted?    One in which I can actually figure out how to deploy it?

it shows this issue has already been reported....  OK,  so can anyone please give me SOME workaround?    I'm already delayed by 3 weeks on this project I've been working on.

Can you tell me the name of the developer actually working on this bug,  and how I can reach them,  I will be happy to help them debug it....   PLEASE PLEASE....HELP...  ME

So Ralph,  this problem has NOT been solved,  and would appreciate it if I can have a one-on-ine conversation with you to help me out.

I will be willing to pay for your time (as long as it's reasonable).   Or do you have other workarounds posted?

Anyway, as far as I'm concerned this issue is NOT closed.

jdcrunchman, 2019-09-20:
RalphHogenbirk  - can you please reach out to me and help me get my Dev system running again.
Please contact me via Signal,  Telegram,  WhatsApp,  or even POTS at the following phone number,  or email me at jdcrunchman at protonmail dot com or jdcrunchman at gmail dot com.

isn't there ANY workaround?

pedroknup, 2019-11-27:
Hi guys. A new version (1.5.0) has been released yesterday that might fix all the compatibility issues from Atom 1.35.0 to 1.42.0 (nightly). Please try it out and drop a message here saying how it went.

closed 2020-02-18 by pedroknup