pycom/pymakr-atom #133: Pysense and pytrack failed to upload with compression

AbrehetIot, 2019-09-02:
Hello
I am not able to upload anything to pysense and pytrack. The problem is, it stops uploading when it reaches some steps **with compression.** Please look the screenshot below how it stops uploading. 
[image]

RalphHogenbirk, 2019-09-02:
Hi @AbrehetIot, very likely your board doesn't support compression. It only works for firmware 1.20 and up. In global settings, disable 'fast upload'. De description of that setting will give you more info :)

AbrehetIot, 2019-09-02:
Thanks more dear@RalphHogenbirk, it is solved now.

closed 2019-09-02 by RalphHogenbirk

bkomac, 2021-01-01:
Yea, on ESP8266 this defiantly does the trick. Tnx!

`"fast_upload": false`