pycom/pymakr-atom #134: Uncaught Error: Could not locate the bindings file. Tried: → $ATOM_HOME/packages/pymakr/node_mod...

PDWest, 2019-09-08:
[Enter steps to reproduce:]

I'm sure this has everything to do with my exact linux version (Mint Rosa 17.3), Atom (1.40.1) and PyMakr (just upgraded (as was Atom: maybe this caused my problem)).

1. Restart Atom
2. Observe red boxes in upper right corner

**Atom**: 1.40.1 x64
**Electron**: 3.1.10
**OS**: Linux Mint
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.15

### Stack Trace

Uncaught Error: Could not locate the bindings file. Tried:
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/Debug/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/Release/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/out/Debug/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/Debug/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/out/Release/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/Release/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/default/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/compiled/10.2.0/linux/x64/bindings.node

```
At /home/phil/.atom/packages/pymakr/node_modules/bindings/bindings.js:96

Error: Could not locate the bindings file. Tried:
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/Debug/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/Release/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/out/Debug/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/Debug/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/out/Release/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/Release/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/default/bindings.node
 → /home/phil/.atom/packages/pymakr/node_modules/serialport/compiled/10.2.0/linux/x64/bindings.node
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:93:9)
    at /packages/pymakr/node_modules/serialport/node_modules/@serialport/bindings/lib/linux.js:1:191)
    at /packages/pymakr/node_modules/serialport/node_modules/@serialport/bindings/lib/linux.js:104:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/serialport/node_modules/@serialport/bindings/lib/index.js:14:22)
    at /packages/pymakr/node_modules/serialport/node_modules/@serialport/bindings/lib/index.js:17:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/serialport/lib/index.js:2:17)
    at /packages/pymakr/node_modules/serialport/lib/index.js:17:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/lib/connections/pyserial.js:5:18)
    at /packages/pymakr/lib/connections/pyserial.js:7:30)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.4.15 
```

araujo-david, 2019-09-09:
I have the same error on Manjaro Linux since I upgraded to version 1.4.15.

`[Enter steps to reproduce:]

1. Upgrade to version 1.4.15
2. Restart Atom

**Atom**: 1.39.1 x64
**Electron**: 4.2.10
**OS**: "Manjaro Linux"
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.15

### Stack Trace

Uncaught Error: Could not locate the bindings file. Tried:
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/build/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/build/Debug/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/build/Release/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/out/Debug/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/Debug/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/out/Release/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/Release/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/build/default/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/compiled/10.11.0/linux/x64/bindings.node

```
At /home/k72dr/.atom/packages/pymakr/node_modules/bindings/bindings.js:96

Error: Could not locate the bindings file. Tried:
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/build/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/build/Debug/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/build/Release/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/out/Debug/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/Debug/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/out/Release/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/Release/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/build/default/bindings.node
 → /home/k72dr/.atom/packages/pymakr/node_modules/serialport/compiled/10.11.0/linux/x64/bindings.node
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:93:9)
    at /packages/pymakr/node_modules/serialport/node_modules/@serialport/bindings/lib/linux.js:1:191)
    at /packages/pymakr/node_modules/serialport/node_modules/@serialport/bindings/lib/linux.js:104:3)
    at Module._compile (/usr/lib/atom/src/native-compile-cache.js:121:30)
    at Object.value [as .js] (/usr/lib/atom/src/compile-cache.js:255:23)
    at Module.load (internal/modules/cjs/loader.js:602:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
    at Function.Module._load (internal/modules/cjs/loader.js:533:3)
    at Module.require (internal/modules/cjs/loader.js:640:17)
    at require (/usr/lib/atom/src/native-compile-cache.js:68:27)
    at /packages/pymakr/node_modules/serialport/node_modules/@serialport/bindings/lib/index.js:14:22)
    at /packages/pymakr/node_modules/serialport/node_modules/@serialport/bindings/lib/index.js:17:3)
    at Module._compile (/usr/lib/atom/src/native-compile-cache.js:121:30)
    at Object.value [as .js] (/usr/lib/atom/src/compile-cache.js:255:23)
    at Module.load (internal/modules/cjs/loader.js:602:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
    at Function.Module._load (internal/modules/cjs/loader.js:533:3)
    at Module.require (internal/modules/cjs/loader.js:640:17)
    at require (/usr/lib/atom/src/native-compile-cache.js:68:27)
    at /packages/pymakr/node_modules/serialport/lib/index.js:2:17)
    at /packages/pymakr/node_modules/serialport/lib/index.js:17:3)
    at Module._compile (/usr/lib/atom/src/native-compile-cache.js:121:30)
    at Object.value [as .js] (/usr/lib/atom/src/compile-cache.js:255:23)
    at Module.load (internal/modules/cjs/loader.js:602:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
    at Function.Module._load (internal/modules/cjs/loader.js:533:3)
    at Module.require (internal/modules/cjs/loader.js:640:17)
    at require (/usr/lib/atom/src/native-compile-cache.js:68:27)
    at /packages/pymakr/lib/connections/pyserial.js:5:18)
    at /packages/pymakr/lib/connections/pyserial.js:7:30)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.4.15 
```

`

nicolsc, 2019-09-09:
Got the same on MacOS

Looks like installing the `@serialport/bindings`  and adding a symlink in .atom/packages/pymakr/node_modules/serialport/ fixes it

araujo-david, 2019-09-10:
It worked for me, thanks.

ainjayla, 2019-09-10:
@nicolsc how do you install @serialport/bindings?

frahlg, 2019-09-11:
This needs to be fixed!

I just got it working rolling back Electron to 4.2.5.

`
npm install -g prebuild
npm install -g prebuild-install
cd ~/.atom/packages/pymakr/node_modules/\@serialport/bindings
prebuild-install --runtime electron --target 4.2.5 --tag-prefix @serialport/bindings@ --verbose
`

RalphHogenbirk, 2019-09-13:
I've released a new fix, that should handle both the linux and the osx issue. Let me know if this solves your issue @PDWest @araujo-david @nicolsc

frahlg, 2019-09-14:
I upgraded, and the same problem occurred as before. I then reinstalled the plugin, with the same result. It seems that it cant locate the bindings file.

Running MacOS 10.14.6 (Mojave). Newest Atom 1.40.1, and the new PyMakr plugin 1.4.16.

[image]

RedNicStone, 2019-09-14:
> Same on Fedora 28, since update to 1.40.0:

**Atom**: 1.40.1 x64
**Electron**: 3.1.10
**OS**: "Fedora release 28"
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.16

### Stack Trace

Uncaught Error: Could not locate the bindings file. Tried:
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/build/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Release/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/Debug/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Release/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/Release/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/build/default/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/compiled/10.2.0/linux/x64/bindings.node

```
At /home/account/.atom/packages/pymakr/node_modules/bindings/bindings.js:96

Error: Could not locate the bindings file. Tried:
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/build/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Release/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/Debug/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Release/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/Release/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/build/default/bindings.node
 → /home/account/.atom/packages/pymakr/node_modules/@serialport/bindings/compiled/10.2.0/linux/x64/bindings.node
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:93:9)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/linux.js:1:191)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/linux.js:104:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:14:22)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:17:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/serialport/lib/index.js:2:17)
    at /packages/pymakr/node_modules/serialport/lib/index.js:17:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/lib/connections/pyserial.js:5:18)
    at /packages/pymakr/lib/connections/pyserial.js:7:30)
```

### Commands

```
```

[... 33 lines cut ...]
    at Module.require (/app.asar/static/index.js:72:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146745)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/linux.js:1:191)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/linux.js:104:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/usr/share/atom/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:14:22)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:17:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147429)
[... cut ...]
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (/usr/share/atom/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/serialport/lib/index.js:2:17)
    at /packages/pymakr/node_modules/serialport/lib/index.js:17:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/static/<embedded>:11:147429)
```

PDWest, 2019-09-15:
I just upgraded to pymakr 1.4.16 and I do not believe that either problem is solved.  When I start Atom, I get two red popup windows.  The first states:
"there was an error with your serialport module, Pymakr will likely not work properly.  Please try to install again or report an issue on our github..."
THIS POPUP Quickly disappears, but a second red popup box persists:
Uncaught Error: Could not locate the bindings file. Tried: → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/bindings.node → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/Debug/bindings.node → /home/phil/.atom/packages/pymakr/node_modules/serialport/build/Release/bindings.node...

QUESTION:  
1) The first message asks me to "Please try to install again", but it doesn't say what it wants me to install again.  Pymakr, Atom, the serial port module--what should I do to pursue this suggestion?

2) Can I do anything about the bindings issue?  

I really need some suggestions about how to get gong again--I'm dead in the water here...  Is there any way I can roll back?

PDWest, 2019-09-15:
As suggested by frahlg I tried to roll back to Electron to 4.2.5.  The `npm install -g prebuild` command failed with a series of  `Error: CERT_UNTRUSTED` messages, but I solved that with a `npm config set strict-ssl false` (kinda hate to do things like that...).  With that, the downloads completed, but the:
`prebuild-install --runtime electron --target 4.2.5 --tag-prefix @serialport/bindings@ --verbose`
command failed with:
`/usr/bin/env: node: No such file or directory`

tkrodgers, 2019-09-16:
Like PDWest, I've done re-installs of Atom & pymakr 1.4.16, and the problem persists. Not an Atom (or PyCom) expert, so very much stuck in the water. Mac OS 10.14.6 (Mojave) here. Nicolsc (above) indicated: "Looks like installing the @serialport/bindings and adding a symlink in .atom/packages/pymakr/node_modules/serialport/ fixes it", but I'm not sure how to either. Happy to go that route until a more elegant fix is in place, but could us help in knowing exactly what commands to run/where.

nicolsc, 2019-09-16:
Hi @tkrodgers 

Following should do the trick.
```
cd ~/.atom/packages/pymakr/
npm install @serialport/bindings
ln -s ~/.atom/packages/pymakr/node_modules/@serialport/bindings ~/.atom/packages/pymakr/node_modules/serialport/bindings
```

PDWest, 2019-09-17:
I completely removed atom, reinstalled atom, installed pymakr, and all appears to be working!

-Phil

jdcrunchman, 2019-09-20:
I did the same thing that PDWest tried...   I installed the new Atom (had to do this first)....
rm -rf ~/.atom and 
~/Library/Preferences/com.github.atom.plist 

Restarted the new Atom 1.40.1,  and re-installed PyMakr 1.4.16,  and still get this....
[image]
 So PDWest,  what did you do that was different then what I did above?
PDWest,  can you please contact me directly at jdcrunchman at protonmail dot com?
or call me at +1 818-640-5290 on Signal, Telegram,  or WhatsApp?

Best time to contact me is 10 am - 5 pm PDT,  or after 7 pm if you're in EU or UK.
There HAS to be some workaround,   isn't there?

jdcrunchman, 2019-09-20:
Nicolsc - I tried your fix...
cd ~/.atom/packages/pymakr/,  but when I tried the other command....
$ npm install @serialport/bindings
-bash: npm: command not found

I CANNOT execute this...   how do I install npn?    Use Apt get?

I tried that,  and got this message.....  $ npm install @serialport/bindings
-bash: npm: command not found

tried the apt get,  and got....   "To use the apt command-line tool you need to install a JDK"
So,  please tell me how I can install npm on my machine so I can try your fix?

jdcrunchman, 2019-09-20:
PDWest,  you never specified what version of Atom and PyMakr you installed....  are they the same as mine?

John

jdcrunchman, 2019-09-20:
Ok,  I did the following fix by Nicolsc but it STILL didn't work.   I had to first install HomeBrew,  then Node from that...  so $ npm -v
6.11.3

then I did this.....  as per your instructions....  

ln -s ~/.atom/packages/pymakr/node_modules/@serialport/bindings ~/.atom/packages/pymakr/node_modules/serialport/bindings
 No error messages.    

I still get the same results,   failure to load the PyMakr with the usual orange dialog box,  so there is just something you are leaving out....  WHAT IS IT?

I've followed every single workaround,  and NONE OF THEM WORK - PLEASE PLEASE - I really have to get this IDE to work.

I will be willing to host a Zoom video conference,  and invite both the Atom,  and PyMakr developers to meet in a Zoom room and iron out this very very long issue....

So,  what ELSE can I try ?

John

nicolsc, 2019-09-20:
Hi @jdcrunchman

NPM is a package management tool used by nodejs devs.
As the Atom packages are nodejs applications, I assumed node and npm were
required. I must admit I never looked closely at how Atom works

Don’t bother installing npm just to get that package
You can just download the @serialport/bindings package at the origin :
https://github.com/serialport/node-serialport?files=1

Then create a symlink, so that Pymakr “find” something when looking into
one of the paths mentioned in the loading error msg

PDWest, 2019-09-20:
@jdcrunchman

IIRC, to remove atom, I followed the first two suggestions on 
https://stackoverflow.com/questions/23621181/how-to-uninstall-atom-text-editor-on-linux
namely:
```
sudo rm /usr/local/bin/atom
sudo rm /usr/local/bin/apm
rm -rf ~/atom
rm -rf ~/.atom
rm -rf ~/.config/Atom-Shell
sudo rm -rf /usr/local/share/atom/
```
and then, for good measure:
`sudo apt-get remove atom`

Then I did a fresh Atom install (I don't specifically recall how, but I think just an apt-get -install), and then installed pymakr from within Atom.
-Phil

jdcrunchman, 2019-09-20:
I was able to get that far,  I have Atom 1.40.1 installed now,  but still cannot get PyMakr installed.   Can you give me some Screen share time,   i can show you exactly what I did.

Do you have a preferred Chat channel?   And can you use Zoom.us?   or better yet,  just text me on Signal,  Telegram,  WhatsApp.

+1 818-640-5290

John

jdcrunchman, 2019-09-20:
nicolsc - I did all of that already....  but that still doesn't work.   Is there any chance I can get a one-on-one interaction with you,   either via voice,  zoom,  or some other voip app like Skype,   Hangouts,  Facebook,  or Signal?

My number is +1 818-640-5290 - call me either on Signal,  Telegram,  or WhatsApp please.

John

PDWest, 2019-09-20:
@jdcrunchman
Did remove atom as you said you did in https://github.com/pycom/pymakr-atom/issues/134#issuecomment-533690287, or as I did in https://github.com/pycom/pymakr-atom/issues/134#issuecomment-533721562 ?

jdcrunchman, 2019-09-20:
PDWest - Yes,   I removed all traces of Atom App,  but kept BOTH the Zip files for 1.40.1 and 1.38.2
Right now,   I have Atom 1.40.1 X64 installed.   I did follow ALL suggested workarounds as was posted in this forum.    Still doesn't work....   I really need hand holding and a screen sharing session of someone is nice enough to talk to me directly....   PLEASE PLEASE text me at +1 818-640-5290
Signal,  Telegram,  or WhatsApp will get to me.

John

jdcrunchman, 2019-09-20:
Is there anything ELSE I should try?

or can someone be nice enough to reach me directly...

John

pwoehrer, 2019-09-21:
Just ran into the same problem with Atom 1.39.1 x64 on Manjaro Linux.
Pymakr is 1.4.16.

It's a bummer because I can't work on my project anymore.

Tugzrida, 2019-09-21:
Same thing's happened to me with Atom 1.40.1 on macOS 10.14.6. This is my fix after doing a fresh re-install of pymakr. I have no idea if this is the best way but it works for me.

    cd ~/.atom/packages/pymakr/
    npm install @serialport/bindings
    mkdir ~/.atom/packages/pymakr/node_modules/serialport/build/
    ln -s ~/.atom/packages/pymakr/node_modules/\@serialport/bindings/build/Release/bindings.node ~/.atom/packages/pymakr/node_modules/serialport/build/bindings.node

PDWest, 2019-09-21:
I'm back at my linux box now, and I actually found the terminal session where I removed atom, pymakr and the serial port bindings, and then reinstalled. Some of the history is missing, but here's most of it:
``
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules $ cd @serialport/
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/@serialport $ ls
binding-abstract  binding-mock  bindings  parser-byte-length  parser-cctalk  parser-delimiter  parser-readline  parser-ready  parser-regex  stream
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/@serialport $ ls -al
total 56
drwxr-xr-x  12 phil phil  4096 Sep 15 15:31 .
drwxr-xr-x 233 phil phil 12288 Sep 16 20:10 ..
drwxr-xr-x   3 phil phil  4096 Sep 15 15:31 binding-abstract
drwxr-xr-x   3 phil phil  4096 Sep 15 15:31 binding-mock
drwxr-xr-x   7 phil phil  4096 Sep 15 15:31 bindings
drwxr-xr-x   2 phil phil  4096 Sep 15 15:31 parser-byte-length
drwxr-xr-x   2 phil phil  4096 Sep 15 15:31 parser-cctalk
drwxr-xr-x   2 phil phil  4096 Sep 15 15:31 parser-delimiter
drwxr-xr-x   2 phil phil  4096 Sep 15 15:31 parser-readline
drwxr-xr-x   2 phil phil  4096 Sep 15 15:31 parser-ready
drwxr-xr-x   2 phil phil  4096 Sep 15 15:31 parser-regex
drwxr-xr-x   3 phil phil  4096 Sep 15 15:31 stream
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/@serialport $ cd bindings
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/@serialport/bindings $ ls
bin  binding.gyp  bindings  build  CHANGELOG.md  lib  LICENSE  node_modules  package.json  README.md  src
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/@serialport/bindings $ ls 0al
ls: cannot access 0al: No such file or directory
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/@serialport/bindings $ ls -al
total 64
drwxr-xr-x  7 phil phil  4096 Sep 15 15:31 .
drwxr-xr-x 12 phil phil  4096 Sep 15 15:31 ..
drwxr-xr-x  3 phil phil  4096 Sep 15 15:31 bin
-rw-r--r--  1 phil phil  1241 Sep 15 15:31 binding.gyp
lrwxrwxrwx  1 phil phil    66 Sep 15 15:31 bindings -> /home/phil/.atom/packages/pymakr/node_modules/@serialport/bindings
drwxr-xr-x  3 phil phil  4096 Sep 15 15:31 build
-rw-r--r--  1 phil phil  3319 Sep 15 15:31 CHANGELOG.md
-rw-r--r--  1 phil phil 10244 Sep 15 15:31 .DS_Store
drwxr-xr-x  2 phil phil  4096 Sep 15 15:31 lib
-rw-r--r--  1 phil phil  1105 Sep 15 15:31 LICENSE
drwxr-xr-x  4 phil phil  4096 Sep 15 15:31 node_modules
-rw-r--r--  1 phil phil  2635 Sep 15 15:31 package.json
-rw-r--r--  1 phil phil   451 Sep 15 15:31 README.md
drwxr-xr-x  2 phil phil  4096 Sep 15 15:31 src
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/@serialport/bindings $ cd ..
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/@serialport $ cd ..
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules $ cd serialport/
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/serialport $ ls -al
total 100
drwxr-xr-x   5 phil phil  4096 Sep 15 15:31 .
drwxr-xr-x 233 phil phil 12288 Sep 16 20:10 ..
drwxr-xr-x   2 phil phil  4096 Sep 15 15:31 bin
lrwxrwxrwx   1 phil phil    66 Sep 15 15:31 bindings -> /home/phil/.atom/packages/pymakr/node_modules/@serialport/bindings
-rw-r--r--   1 phil phil 34449 Sep 15 15:31 CHANGELOG.md
drwxr-xr-x   2 phil phil  4096 Sep 15 15:31 lib
-rw-r--r--   1 phil phil  1105 Sep 15 15:31 LICENSE
drwxr-xr-x   3 phil phil  4096 Sep 15 15:31 node_modules
-rw-r--r--   1 phil phil  2763 Sep 15 15:31 package.json
-rw-r--r--   1 phil phil 19570 Sep 15 15:31 README.md
-rw-r--r--   1 phil phil   230 Sep 15 15:31 test.js
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/serialport $ cd bindings
(MachineLearning) phil@phils-i5 ~/.atom/packages/pymakr/node_modules/serialport/bindings $ ls -al
total 64
drwxr-xr-x  7 phil phil  4096 Sep 15 15:31 .
drwxr-xr-x 12 phil phil  4096 Sep 15 15:31 ..
drwxr-xr-x  3 phil phil  4096 Sep 15 15:31 bin
-rw-r--r--  1 phil phil  1241 Sep 15 15:31 binding.gyp
lrwxrwxrwx  1 phil phil    66 Sep 15 15:31 bindings -> /home/phil/.atom/packages/pymakr/node_modules/@serialport/bindings
drwxr-xr-x  3 phil phil  4096 Sep 15 15:31 build
-rw-r--r--  1 phil phil  3319 Sep 15 15:31 CHANGELOG.md
-rw-r--r--  1 phil phil 10244 Sep 15 15:31 .DS_Store
drwxr-xr-x  2 phil phil  4096 Sep 15 15:31 lib
-rw-r--r--  1 phil phil  1105 Sep 15 15:31 LICENSE
drwxr-xr-x  4 phil phil  4096 Sep 15 15:31 node_modules
[... 31 lines cut ...]
Do you want to continue? [Y/n] y
(Reading database ... 252634 files and directories currently installed.)
Removing atom (1.40.1) ...
Processing triggers for desktop-file-utils (0.22-1ubuntu1.1) ...
Processing triggers for gnome-menus (3.10.1-0ubuntu2) ...
Processing triggers for mime-support (3.54ubuntu1.1) ...
(MachineLearning) phil@phils-i5 ~ $ sudo apt-get install atom
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  atom
0 upgraded, 1 newly installed, 0 to remove and 11 not upgraded.
Need to get 0 B/106 MB of archives.
After this operation, 602 MB of additional disk space will be used.
Selecting previously unselected package atom.
(Reading database ... 243056 files and directories currently installed.)
Preparing to unpack .../archives/atom_1.40.1_amd64.deb ...
Unpacking atom (1.40.1) ...
Processing triggers for desktop-file-utils (0.22-1ubuntu1.1) ...
Processing triggers for gnome-menus (3.10.1-0ubuntu2) ...
Processing triggers for mime-support (3.54ubuntu1.1) ...
Setting up atom (1.40.1) ...
``
@jdcrunchman note that my final step is to install atom with an apt-get, not from zip (which you mentioned that you had maintained)

jdcrunchman, 2019-09-21:
Have you done a parity check to see whats different between unzipping the archive vs using "Apt get"
Also,   if you were sucessful using Apt Get instead of unzipping the archive,  can you share with us the arguments you used in your "apt get"?     Whats the link please?

John

jdcrunchman, 2019-09-21:
PDWest - is there anyway I can contact you directly,  because although these instructions are long,  in my situation,  there is always going to be places when I do that,  I'm going to encounter something completely different.

John

PDWest, 2019-09-21:
> Have you done a parity check to see whats different between unzipping the archive vs using "Apt get"

Nope: I've really got no idea how to go about doing something like that...

> Also, if you were sucessful using Apt Get instead of unzipping the archive, can you share with us the arguments you used in your "apt get"? Whats the link please?
> 
> John

Sure, that about 18 lines from the bottom of the dialog that I posted:
(MachineLearning) phil@phils-i5 ~ $ sudo apt-get install atom

jdcrunchman, 2019-09-21:
Great,  you've been a great help,   but the problem I see from all these tries,   os that there is some sort of "Binding file" issue.   Not sure if it was complaining whether or not this file exists,   or the wrong type,   and specific versions of this file would work,   or it's in the wrong place where Atom expects to find it.

This was displayed in my orange error message dialog box I got when trying to install PyMakr.

/Users/johndraper/.atom/packages/pymakr/node_modules/bindings/bindings.js:96

Let me ask you this (and other Atom developers listening),   are all essential files located in the ~/.atom directory?

This line was displayed after the error displayed all the paths it tried to look for before it actually gave up,   is that what the dialog was saying?

But then,  this other file as specified in the path I gave above,  called "bindings.js:96" is that the file where the code was trying to find this binding file?

Its probably going to take me about 4 - 5 hours trying your (PDWest ) fix,  using your shell dialog you sent earlier.
You also mentioned it wasn't complete,   so can I contact you privately on some of this stuff,  just because I'm concerned about sending file paths over a public forum.

Just an email address for you would be helpful,  just send contact details to my jdcrunchman at protonmail dot com.   Would be appreciated.    I also have (and use) Zoom on certain occasions,  with my camera disabled,  used to share my screen,   which in most situations works really well,  because MOST people don't have MY problems,   due to differences in the stuff I have on my machine,  like Anaconda,  where normal file pathways to things like PyCharm are totally different.

I've been avoiding PyCharm,  because of it's expense and also at the time I researched it,  PyCharm had no support for ESP32 development,  but my friend Andy says they now support ESP32,  but I personally have not checked into it.  Then,  what about PyCom (a commercial business - selling pre-made LoRa hardware,  and I think is the source of PyMakr.   What is their relationship with Atom,  MicroPython, and PyMakr?

I have spent at least 250 hours total in trying to deal with the very unstable IDE as well as MicroPython,  MicroWebSrv etc,   I just don't think I deserved to spend more then 200 hours over 6- 8 months on a very unstable development system,   so now my supporters have now realized it's time to switch to an enterprise version of an IDE....   But my observation seems to be that we have 3 main developers.   Atom, PyMakr,  and Micropython with apparently (and I say Apparently) very little reaction and cooperation among them,   then we have PyCom developers but they have their own commercial interests and apparently support open source.

I should have finished this project before Burning man,  which would have been a wonderful opportunity to really do good field testing of our device.

Anyway,   I hope that all of you developers would get a little better cooperation in working on this very long issue which has been in the GitHub issue list forever practically.

Allthough I'm getting fairly fast response to my questions, issues, and concerns here,  there is a bit of privacy issues I'm having to deal with that my supports have brought to my attention,   so upon initial contact with people in this forum,  I really would like to have emails and contact info for the following developers so I can somewhat control who gets to see paths on my personal machine.

I already have MIcropython and MicrowebSrv developers I'm talking to directly have resulted in BOTH of us benefitting,  mostly due to the uniqueness of my Mac OS laptop which was configured to work with Anaconda,   requiring things like IDE's being setup much different then without it.

But getting a contact number (Signal, Email, or Telegram) for PyMakr,  and Atom would allow me to go directly to the "Horses mouth" as well as protect MY privacy where there are things I have to decide to stay off the public forums and more with one-on-one more private discussions directly with the developers.   I'm sure they are going to want to have their stuff run on everyone's computer.

As well as the main person in charge with Atom,  or at least the project leader,  as I'm assuming there must be huge numbers of volunteers so far working on Atom.   So right now,  would like to know who the Pymakr developer is.

Anyway, these are my thoughts,  and all I ask is for someone to reach out to me and talk to me,  not just because I have a specific problem,  but the solutions of which could help everyone using Atom/PyMakr.

RalphHogenbirk is one developer I would like to talk to,  primarily because looking at his Bio is a drone racer,  and works in part with PyCom.   I have no direct way to reach him.

Thanx

John

jdcrunchman, 2019-09-21:
So,  in conclusion to this whole mess,  it just involves inserting a file into a specific folder?    Or is it a lot more complex?    Whats the file?    Where does it go?

Tugzrida, 2019-09-22:
Yes @jdcrunchman, the temporary fix is really quite simple.

If you open the stack trace in the second red bubble, at the beginning there's a bunch of lines beginning with an arrow. These are the paths at which it is expecting a `bindings.node` file.

My solution a few posts ago works reliably for me on macOS to install and then link the bindings file to one of these expected paths. I don't know if it will work forever, but it will get you up and running and hopefully by the next update it will all just work.

You may also want to try VS Code instead of Atom - I haven't used it much but it's another option.

pwoehrer, 2019-09-22:
Tried linking the missing file in one of the directories of the trace. Did not work. Giving up.

Thanks for all the suggestions.

jdcrunchman, 2019-09-22:
I did the same thing,  didn't work for me at all.    I saw your post to the
forum with your console dialog.   Haen't tried it yet,  since I believe you
weren't using a Mac and thus might have different conditions.    Is there
anyway you would consent to a screen sharing session,  giving you a chance
to "Look around" to see something you might not know?

Your handle in that forum us PDWest,   right?    Is any new planned update
scheduled soon that addresses this?    I know you are super busy,  but can
I at least ask you for a screen sharing session?

John

jdcrunchman, 2019-09-22:
Cameron,   I hope this reply goes to you,   if not,   I'll try reposting
this to the forum in the hopes of finding your email address so I can talk
to you directly instead of flooding the group.    I tried that method above
that you mentioned,  it did NOT work for me,  and it appears the ONLY way I
can fix this is for one of you guys to be nice enough to screen share with
me,    I've given my contact info out to the group MANY TIMES,  but I have
no direct way to reach anyone.

Is there anyway you can call me Via Signal or Telegram?   at +1
818-640-5290?

John

jdcrunchman, 2019-09-22:
PDWest,   I tried to send you a private email through the group,   did you get it?    Again,  I keep asking you for your email address,    why can't you allow me to talk to you without cluttering the forum?

John

jdcrunchman, 2019-09-22:
I looked even more carefully at the stack trace,   I repeat it here,  so I can point out a few questions about it.   Please don't Dis me for being lame,  but I do have a learning disability and asking for a little more "hand holding" if anyone is nice enough,  so in the meantime,  I'm going to continue posting what I'm doing in the hopes that someone might point out what I'm doing wrong...   SO lets see my stack trace and let me go over it,   asking questions about it.

Error: Could not locate the bindings file. Tried:
 → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/bindings.node
 → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Release/bindings.node
 → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/Debug/bindings.node
 → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Release/bindings.node
 → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/Release/bindings.node
 → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/default/bindings.node
 → 

So,  above dialog is all the places it looked for the bindings file,    RIGHT?   

/Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/compiled/10.2.0/darwin/x64/bindings.node

I'm confused about these following lines:    
    at bindings (/Users/johndraper/.atom/packages/pymakr/node_modules/bindings/bindings.js:93:9)
    at Object.<anonymous> 

What does this previous line mean....?

(/Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/darwin.js:1:191)
    at Object.<anonymous> 

I have NO CLUE what this means!!!   can someone explain it to me?

(/Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/darwin.js:104:3)

What is this file path above,   and what are they trying to tell me?
Then all these pathways below,  what do they mean,  and why are they mentioned in the stack trace?

    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:72:46)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146745)
    at Object.<anonymous> (/Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/index.js:10:22)
    at Object.<anonymous> (/Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/index.js:17:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:72:46)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146745)
    at Object.<anonymous> (/Users/johndraper/.atom/packages/pymakr/node_modules/serialport/lib/index.js:2:17)
    at Object.<anonymous> (/Users/johndraper/.atom/packages/pymakr/node_modules/serialport/lib/index.js:17:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:72:46)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:146745)
    at Object.<anonymous> (/Users/johndraper/.atom/packages/pymakr/lib/connections/pyserial.js:5:18)
    at Object.<anonymous> (/Users/johndraper/.atom/packages/pymakr/lib/connections/pyserial.js:7:30)

I tried looking at the code in these files,   I totally do NOT understand it.   Can someone please explain it to me?

John

PDWest, 2019-09-22:
> PDWest, I tried to send you a private email through the group, did you get it? Again, I keep asking you for your email address, why can't you allow me to talk to you without cluttering the forum?
> 
> John

Did I get the private email?  I don't know  I was unaware this site had that.  The last thing I want is another email address...

why can't I allow you to talk to you without cluttering the forum? Because I'm a nearly clueless end-user who just used the  linux GUI search function to find and then delete anything he could find relating to atom, pymakr or nodemodules, reinstalled, and was fortunate enough to have that work.  Plus, if any info I might provide would help you, it might help someone else, too, if posted here.

jdcrunchman, 2019-09-22:
Here is my comment to your post Cameron...

> Yes @jdcrunchman <https://github.com/jdcrunchman>, the temporary fix is
> really quite simple.
Simple for YOU but certainly not simple for me.    I still don't understand
this....  I respectfully request a Screen sharing session if you are kind
enough to help me,   I just need your email address and contact
information.

> If you open the stack trace in the second red bubble, at the beginning
> there's a bunch of lines beginning with an arrow. These are the paths at
> which it is expecting a bindings.node file.
What is the second red bubble?    Can you just send me the file and can't I
just copy it into the respective path?    or is your file somehow
completely different then one I can generate?

Tugzrida posted this:

cd ~/.atom/packages/pymakr/
npm install @serialport/bindings
mkdir ~/.atom/packages/pymakr/node_modules/serialport/build/

ln -s
~/.atom/packages/pymakr/node_modules/\@serialport/bindings/build/Release/bindings.node
~/.atom/packages/pymakr/node_modules/serialport/build/bindings.node

I tried this,  but it didn't work.

> My solution a few posts ago works reliably for me on macOS to install and
> then link the bindings file to one of these expected paths. I don't know if
> it will work forever, but it will get you up and [...]
Here is my directory traversal....  and as you can see,  these files DO
exist....  right in front of you....

(base) Johns-MacBook-Pro-4:.atom johndraper$ cd packages

These are the files in my packages directory...
(base) Johns-MacBook-Pro-4:packages johndraper$ ls -als
total 0
0 drwxr-xr-x   3 johndraper  staff  102 Sep 20 23:06 .
0 drwxr-xr-x  11 johndraper  staff  374 Sep 20 13:52 ..
0 drwxr-xr-x  18 johndraper  staff  612 Sep 20 23:06 pymakr
(base) Johns-MacBook-Pro-4:packages johndraper$ cd pymakr
(base) Johns-MacBook-Pro-4:pymakr johndraper$ ls -als
total 272
  0 drwxr-xr-x   18 johndraper  staff     612 Sep 20 23:06 .
  0 drwxr-xr-x    3 johndraper  staff     102 Sep 20 23:06 ..
  0 drwxr-xr-x    5 johndraper  staff     170 Sep 20 23:06 .github
  8 -rw-r--r--    1 johndraper  staff      59 Sep 20 23:06 .npmignore
 16 -rw-r--r--    1 johndraper  staff    8052 Sep 20 23:06 CHANGELOG.md
  8 -rw-r--r--    1 johndraper  staff     285 Sep 20 23:06 LICENSE.md
 16 -rw-r--r--    1 johndraper  staff    6952 Sep 20 23:06 README.md
  8 -rw-r--r--    1 johndraper  staff       2 Sep 20 23:06
connection_state.json
  0 drwxr-xr-x    3 johndraper  staff     102 Sep 20 23:06 keymaps
  0 drwxr-xr-x   12 johndraper  staff     408 Sep 20 23:06 lib
  0 drwxr-xr-x    3 johndraper  staff     102 Sep 20 23:06 menus
  0 drwxr-xr-x  232 johndraper  staff    7888 Sep 20 23:06 node_modules .
       <----- as you see here,  node_modules is present.
208 -rw-r--r--    1 johndraper  staff  105324 Sep 20 23:06 package-lock.json
  8 -rw-r--r--    1 johndraper  staff    3044 Sep 20 23:06 package.json
  0 drwxr-xr-x    6 johndraper  staff     204 Sep 20 23:06 precompiles
  0 drwxr-xr-x    8 johndraper  staff     272 Sep 20 23:06 scripts
  0 drwxr-xr-x    4 johndraper  staff     136 Sep 20 23:06 spec
  0 drwxr-xr-x    5 johndraper  staff     170 Sep 20 23:06 styles
(base) Johns-MacBook-Pro-4:pymakr johndraper$ cd node_modules .
 <------ so I go into THAT directory.
(base) Johns-MacBook-Pro-4:node_modules johndraper$ ls -als
total 0
0 drwxr-xr-x  232 johndraper  staff  7888 Sep 20 23:06 .
0 drwxr-xr-x   18 johndraper  staff   612 Sep 20 23:06 ..
0 drwxr-xr-x   19 johndraper  staff   646 Sep 20 23:06 .bin
0 drwxr-xr-x   12 johndraper  staff   408 Sep 20 23:06 @serialport .
  <----- here is ONE "serialport",  but what does that "@" mean?
0 drwxr-xr-x    9 johndraper  staff   306 Sep 20 23:06 ajv
0 drwxr-xr-x    8 johndraper  staff   272 Sep 20 23:06 semver

<<< deleted a lot of stuff I didnt think was important>>>>
0 drwxr-xr-x   11 johndraper  staff   374 Sep 20 23:07 serialport .
    <----- and here is another serialport diretory.
0 drwxr-xr-x    7 johndraper  staff   238 Sep 20 23:06 set-blocking

<<< more stuff deleted >>>>
(base) Johns-MacBook-Pro-4:node_modules johndraper$ cd serialport

Now going in the serialport directory...
(base) Johns-MacBook-Pro-4:serialport johndraper$ ls -als
total 144
 0 drwxr-xr-x   11 johndraper  staff    374 Sep 20 23:07 .
 0 drwxr-xr-x  232 johndraper  staff   7888 Sep 20 23:06 ..
72 -rw-r--r--    1 johndraper  staff  34449 Sep 20 23:06 CHANGELOG.md
 8 -rw-r--r--    1 johndraper  staff   1105 Sep 20 23:06 LICENSE
40 -rw-r--r--    1 johndraper  staff  19570 Sep 20 23:06 README.md
 0 drwxr-xr-x    6 johndraper  staff    204 Sep 20 23:06 bin
 8 lrwxr-xr-x    1 johndraper  staff     73 Sep 20 23:07 bindings ->
/Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings

<<< and look at this above!!!    SEE THAT?    So isn't this proof that I
did follow Tugzrida's post....

 0 drwxr-xr-x    4 johndraper  staff    136 Sep 20 23:06 lib
 0 drwxr-xr-x    3 johndraper  staff    102 Sep 20 23:06 node_modules .
  <------ And look at this!!!!   That file is there?   So why isn't this
working?

WHAT ELSE DID I DO WRONG,  and what did I leave out?

 8 -rw-r--r--    1 johndraper  staff   2867 Sep 20 23:06 package.json
 8 -rw-r--r--    1 johndraper  staff    230 Sep 20 23:06 test.js
(base) Johns-MacBook-Pro-4:serialport johndraper$

There HAS to be something else nobody is telling me!!!!   or this would
work.

jdcrunchman, 2019-09-22:
I just posted more info on my problem,  including lists of the files from
the relevant directories...    it is clear from the past message I send
that THESE FILES ARE THERE!!!!

There HAS to be something else I overlooked....   can anyone else see what
it is?

And when will this update come?

John

jdcrunchman, 2019-09-22:
On my MacOS,   the pathway PyMakr is looking into is expecting me to have a "Build" directory as part of the path.    But in my folders,  I have no such "Build" directory....

→ /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/bindings.node

Look above in the path,   when I looked for the "build" directory in any of my paths,   I don't have it,  and thus could not find the bindings files,  because whoever built this for the Mac OS F**ked up,
and didn't put them in the right place.

ALL of these paths are displayed in the error message.    Due to my poor eyesight,  I didn't see the "build" included in the path.

So,   how can I tell PyMakr and give it the right path to search for these bindings files.
So I think this is just a F**kup on the part of the developers,   who created and built these pathways which are totally WRONG!!!

What is an easy fix?     What happens if I just remove the "build" directory,  and complete the paths
Can someone please tell me what files are supposed to be in the "build" directory,  so I can possibly just try to fix this by hand?

How is PyMakr building these directories?   Where does it get the path information?

Is anyone else finding a "build" directory in the paths in questions?

Tugzrida, 2019-09-23:
@jdcrunchman, I don’t mean to sound rude, but one-on-one support is not generally the way open-source projects work. None of us are being paid - I’m just some kid in Australia 😂

That being said, I’d still like to help you, so here’s a more detailed walkthrough.
First, I’m going to get you to purge and re-install pymakr:

    1. Quit Atom
    2. rm -r ~/.atom/packages/pymakr/
    3. Relaunch Atom and install pymakr(Atom>Preferences>Install>Search for pymakr>Install)

Two red error messages should appear after the install has finished. Showing the stack trace in the second one should show a list of tried locations for `bindings.node` with arrows in front of them. For me, they're in subfolders of `~/.atom/packages/pymakr/node_modules/serialport/` (note the missing `@`). Copy and paste these paths somewhere else, you might need to reference them later.

Now:

    1: Quit Atom
    2: cd ~/.atom/packages/pymakr/
    3: npm install @serialport/bindings
    4: find . -iname *bindings.node

This will list all the `bindings.node` files present in the pymakr install. For me, one of the files is `./node_modules/@serialport/bindings/build/Release/bindings.node` (under the `@serialport` folder)

Now we need to link this actual `bindings.node` file to one of the places it's expected to be. As I think you mentioned, you don't have a build directory, so we need to make one before linking the file there:

    1: mkdir ~/.atom/packages/pymakr/node_modules/serialport/build/
    2: ln -s ~/.atom/packages/pymakr/node_modules/\@serialport/bindings/build/Release/bindings.node ~/.atom/packages/pymakr/node_modules/serialport/build/bindings.node

For this last section, you need to make sure that the paths match what's showing up on your machine. Essentially the previous commands are:

    1: mkdir {directory where pymakr is looking for bindings.node, from stack trace}
    2: ln -s {actual bindings.node file from the find command} {where pymakr is looking for bindings.node, from stack trace}

Now you can relaunch Atom and pymakr should load normally.

If the paths from your stack trace or find command differ from mine(and yes the `@serialport` and `serialport` folders are different), then my exact commands will not work. Everything up to the last block of instructions above will be the same, you'll just need to substitute the paths you have for the `mkdir` and `ln -s` commands as I described.

If that still doesn't work for you, then there's not much else I can do. So you'll either need to wait for an update to fix this, or you could use VS Code and its pymakr extension instead, which is free as well.

jdcrunchman, 2019-09-23:
Tugzrida - I mentioned several times that I was willing to pay them for their time in giving me on-on-one help.

But I will first try out your suggestions.

jdcrunchman, 2019-09-23:
Tugzrida - OK,   I'm going through each of your steps..

1. Quit Atom
2. rm -r ~/.atom/packages/pymakr/
3. Relaunch Atom and install pymakr(Atom>Preferences>Install>Search for pymakr>Install)

I got the orange dialog which displayed these pathways.

Uncaught Error: Could not locate the bindings file. Tried: → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/bindings.node → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Debug/bindings.node → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Release/bindings.node → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Debug/bindings.node → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/Debug/bindings.node → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Release/bindings.node → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/Release/bindings.node → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/default/bindings.node → /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/compiled/10.2.0/darwin/x64/bindings.node

you wrote:    " For me, they're in subfolders of ~/.atom/packages/pymakr/node_modules/serialport/ (note the missing @). Copy and paste these paths somewhere else, you might need to reference them later.

YOU SAID:   "note the missing @" - Well,  I have news for you,  in MY error list it is NOT MISSING!!
it is there.    Then,  you said:

>Copy and paste these paths somewhere else, you might need to reference them later.

My question for you....   when copying and pasting these paths,   MY copy has the @,   am I supposed to remove the @ after copying the path and saving it?    Or do I leave the @ in the path?   you did NOT make that clear.   And you said:  "note the missing @",   well,  it's NOT missing.

Next you said "Quit Atom",   then to ussue these shell commands....
cd ~/.atom/packages/pymakr/

Then:    npm install @serialport/bindings

This is what I got after doing that.

> @serialport/bindings@2.0.8 install /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

+ @serialport/bindings@2.0.8
updated 1 package and audited 1481 packages in 3.348s
found 3 vulnerabilities (1 moderate, 2 high)
  run `npm audit fix` to fix them, or `npm audit` for details
(base) Johns-MacBook-Pro-4:pymakr johndraper$ 

Is this what I'm expecting to get?

$ find . -iname *bindings.node
./node_modules/@serialport/bindings/build/Release/bindings.node
./precompiles/serialport-linux/bindings.node
./precompiles/serialport-osx/bindings.node
./precompiles/serialport-win/bindings.node
./precompiles/serialport-win32/bindings.node
(base) Johns-MacBook-Pro-4:pymakr johndraper$ 

You wrote:
>For me, one of the files is ./node_modules/@serialport/bindings/build/Release/bindings.node (under the @serialport folder)

NOTE ABOVE....Only the top one in the list is as you described.

>Now we need to link this actual bindings.node file to one of the places it's expected to be. As I think you mentioned, you don't have a build directory, so we need to make one before linking the file there:

$mkdir ~/.atom/packages/pymakr/node_modules/serialport/build/

So I created the directory as you described.

$ ln -s ~/.atom/packages/pymakr/node_modules/\@serialport/bindings/build/Release/bindings.node ~/.atom/packages/pymakr/node_modules/serialport/build/bindings.node

So,  I linked it as you described...  no errors so far.

You then wrote:
>For this last section, you need to make sure that the paths match what's showing up on your machine. Essentially the previous commands are:
>1: mkdir {directory where pymakr is looking for bindings.node, from stack trace}

I'm confused about this:    did you want me to examine the path (ON MY MACHINE)
so,   on my machine,  I did:    cd /.atom/packages/pymakr/node_modules/serialport
I guess I'm now to confirm that "build" is in the serialport,  is that right?

So I listed the serialport directory and got....
$ ls -als
[... 13 lines cut ...]
(base) Johns-MacBook-Pro-4:serialport johndraper$ pwd
/Users/johndraper/.atom/packages/pymakr/node_modules/serialport

So now I have "build" in my path...  but I'm still confused about this statement you typed....
>1: mkdir {directory where pymakr is looking for bindings.node, from stack trace}

What does this mean....you said "{directory where pymakr is looking for bindings.node, from stack trace}",   did you mean this:

>/Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/bindings.node 

So do I do this?

$$ mkdir /Users/johndraper/.atom/packages/pymakr/node_modules/@serialport/bindings/build/bindings.node

$ln -s ./node_modules/@serialport/bindings/build/Release/bindings.node

Ok,  so I did this,  from the -find comment as you instructed.

Now,  ready to try.
OMG - IT WORKS - for other on this list,   a few things were left out....   

1) Nothing was said that I had to create a "build" directory
2) earlier instructions were incomplete....  THANK YOU THANK YOU Tugzrida for putting these instructions so that a moron like me can understand.

John

jdcrunchman, 2019-09-24:
This fix actually worked.   Earlier fixes published skipped a lot of
important details,  like the fact that I had to create some "build"
directory and put it on the path.
This took me 8 hours to do,   I'm NOT a fast learner.

John

RalphHogenbirk, 2019-09-26:
That was a long read, but there is a potential fix on the develop branch right now. Anyone here who likes to test it?
How to test: 
- Uninstall pymakr
- Download the contents from the develop branch, unzip on your pc
- Run the 'apm install' and 'apm link' commands from that folder
- Reload Atom

I went over all the stuff that was mentioned on the top here. After a ton of testing and very inconsistent results and an random inability to reproduce it... I think the solution was simply to add @serialport/bindings to the package.json. On my pc, it normally just installs it with 'serialport', but adding @serialport/bindings as well, forces this for everyone. It also detects it automatically when installed from the package.json, so no need to create build folders and link the bindings.node file separately.

Let me know if someone tested it, and if it solved the issue!

jdcrunchman, 2019-09-29:
I certainly do NOT want to test it,   I lost more then 2 months on this
project because of this bug.    I have to push on to finish this project.

John

Tugzrida, 2019-09-30:
@RalphHogenbirk Looks like this commit is in 1.4.17 now. I just did a fresh re-install of pymakr to 1.4.17 and I'm still facing the same issue in Atom 1.40.1 on macOS 10.14.6.

    Error: Could not locate the bindings file. Tried:
    ~/.atom/packages/pymakr/node_modules/serialport/build/bindings.node
    ~/.atom/packages/pymakr/node_modules/serialport/build/Debug/bindings.node
    ~/.atom/packages/pymakr/node_modules/serialport/build/Release/bindings.node
    ~/.atom/packages/pymakr/node_modules/serialport/out/Debug/bindings.node
    ~/.atom/packages/pymakr/node_modules/serialport/Debug/bindings.node
    ~/.atom/packages/pymakr/node_modules/serialport/out/Release/bindings.node
    ~/.atom/packages/pymakr/node_modules/serialport/Release/bindings.node
    ~/.atom/packages/pymakr/node_modules/serialport/build/default/bindings.node
    ~/.atom/packages/pymakr/node_modules/serialport/compiled/10.2.0/darwin/x64/bindings.node

[My original bodge](https://github.com/pycom/pymakr-atom/issues/134#issuecomment-533794141) still works. Happy to test anything else :)

elmorg, 2019-10-01:
> That was a long read, but there is a potential fix on the develop branch right now. Anyone here who likes to test it?
> How to test:
>
> * Uninstall pymakr
> * Download the contents from the develop [...]

This worked for me, but I had to reload atom after uninstalling pymakr, then run the apm commands, then reload atom again.

jdcrunchman, 2019-10-01:
By the way,  since I was able to get the files in the ~.atom directory
sorted out,    can I just Zip up these files and pass them to my partner,
since he could just un-stuff them,   into their own ~.atom directory,   and
if not,   why not?

The zip file is too large to pass through my Protonmail or gmail account so
Im going to hav to pass them through a cloud based service.

Do you know if any files that might lock them into only running on my
specific machine.    I'm hoping they use relative paths tree'ing down from
my
~/.atom directory where all path references seem to be derived from.

Because if this works,   I can put them up on the cloud,   and give the
link to others who don't want to go through those 20 steps to sort out
these PyMakr issues.

John

teusH, 2019-10-30:
Used the trick to create a symbolic link as suggested and it worked for me. Platforms: OSx and Linux.
Was using a fresh install of atom as pymakr. The problem maker was pymakr saying in a red window: missing 'bindings.node' (in my case a list of just one file). The list was a list of search path pymakr was trying to find this missing file. I found this file in the precompiled map for many platforms. Creating a sym link from the first path pymakr was complaining about to this precompiled 'bindings.node' did the trick. Eg use 'find ~/.atom/packages/pymakr -name bindings.node' to find the endpoint of the symlink.
Thanks for hinting me to that solution. Yes this type of problems makes you nervous :-)

elmasria, 2021-01-15:
A new version exists that solves the above issue. if you face it again, please reopen the issue.

closed 2021-01-15 by elmasria