pycom/pymakr-atom #137: Uncaught Error: ENOENT, benchmarks/benchmarks not found in ~/Desktop/Atom.app/Contents/Resources/...

Riverj17, 2019-09-22:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.40.1 x64
**Electron**: 3.1.10
**OS**: Mac OS X 10.14
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.16

### Stack Trace

Uncaught Error: ENOENT, benchmarks/benchmarks not found in /Users/JoelRivera/Desktop/Atom.app/Contents/Resources/app.asar

```
At ELECTRON_ASAR.js:112

Error: ENOENT, benchmarks/benchmarks not found in /Users/JoelRivera/Desktop/Atom.app/Contents/Resources/app.asar
    at notFoundError (ELECTRON_ASAR.js:108:19)
    at Object.fs.lstatSync (ELECTRON_ASAR.js:267:9)
    at Sync._getFilesRecursive (/packages/pymakr/lib/board/sync.js:751:22)
    at Sync._getFilesRecursive (/packages/pymakr/lib/board/sync.js:755:33)
    at Sync._getFilesRecursive (/packages/pymakr/lib/board/sync.js:755:33)
    at Sync._getFilesRecursive (/packages/pymakr/lib/board/sync.js:755:33)
    at Sync._getFilesRecursive (/packages/pymakr/lib/board/sync.js:755:33)
    at Sync._getFilesRecursive (/packages/pymakr/lib/board/sync.js:755:33)
    at /packages/pymakr/lib/board/sync.js:252:27
    at end (/packages/pymakr/lib/board/shell.js:261:7)
    at /packages/pymakr/lib/helpers/utils.js:32:11
    at /packages/pymakr/lib/board/shell-workers.js:76:17
    at /packages/pymakr/lib/board/shell.js:416:11
```

### Commands

```
     -5:30.4.0 core:paste (input.hidden-input)
     -5:26.1.0 core:save (input.hidden-input)
```

### Non-Core Packages

```
pymakr 1.4.16 
script 3.18.1 
```

RalphHogenbirk, 2019-09-27:
That looks like a more general issue with your Atom install. Does it only occur when you try to upload with Pymakr code?

elmasria, 2019-10-21:
Hello @Riverj17, What is the status now? 
Have you tried to install the latest version 1.4.17?

Riverj17, 2019-10-21:
After updating to the latest version the error has corrected itself.

closed 2019-10-21 by Riverj17