pycom/pymakr-atom #138: Uncaught Error: error: 193/\?\$ATOM_HOME\packages\pymakr\node_modules\@serialport\bindings\build\...

MrJekyll23, 2019-09-27:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.40.1 ia32
**Electron**: 3.1.10
**OS**: LAPTOP
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.16

### Stack Trace

Uncaught Error: error: 193\\?\C:\Users\Mr.Jack\.atom\packages\pymakr\node_modules\@serialport\bindings\build\Release\bindings.node

```
At C:\Users\Mr.Jack\.atom\packages\pymakr\node_modules\bindings\bindings.js:88

Error: error: 193\\?\C:\Users\Mr.Jack\.atom\packages\pymakr\node_modules\@serialport\bindings\build\Release\bindings.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:146745)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/win32.js:1:191)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:6:22)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/serialport/lib/index.js:2:17)
    at /packages/pymakr/node_modules/serialport/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:147429)
```

### Commands

```
```

### Non-Core Packages

```
atom-material-syntax 1.0.8 
atom-material-ui 2.1.3 
atom-save-all 0.4.1 
autocomplete-gml 0.1.7 
busy-signal 2.0.1 
highlight-selected 0.16.0 
indent-guide-improved 1.4.13 
intentions 1.1.5 
language-gml 0.5.0 
language-javascript-jsx 0.3.7 
linter 2.3.1 
linter-gml 0.1.3 
linter-jshint 3.1.17 
linter-ui-default 1.8.0 
minimap 4.29.9 
minimap-find-and-replace 4.5.2 
minimap-highlight-selected 4.6.1 
pymakr 1.4.16 
semanticolor 3.6.0 
```

MrJekyll23, 2019-09-27:
Whats this about, cann someone help me?

RalphHogenbirk, 2019-09-27:
That's a strange one. There is a bug pending (#134) regarding the serialport bindings, but not this one. First try to remove the whole pymakr package, then install the latest 1.4.17, see if that solves the issue.

elmasria, 2019-10-21:
Hello @MrJekyll23, Have you tried to install the latest version 1.4.17? 
What is the status now?

closed 2020-02-18 by pedroknup