pycom/pymakr-atom #141: Uncaught Error: The specified module could not be found

johnsonthie, 2019-10-24:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Installed Pymaker in Atom
2.
3.

## What happens?

An error message appeared on the top right immediately after the installation was completed. 

## What were you expecting to happen?

No error message

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

Uncaught Error: The specified module could not be found. \?\C:\Users\Johnson.Thie.atom\packages\pymakr\node_modules\@serialport\bindings\build\Release\bindings.node

C:\Users\Johnson.Thie\.atom\packages\pymakr\node_modules\bindings\bindings.js:88
Hide Stack Trace
Error: The specified module could not be found.
\\?\C:\Users\Johnson.Thie\.atom\packages\pymakr\node_modules\@serialport\bindings\build\Release\bindings.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:160:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:722:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:160:31)
    at Module.load (internal/modules/cjs/loader.js:602:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
    at Function.Module._load (internal/modules/cjs/loader.js:533:3)
    at Module.require (file:///C:/Users/Johnson.Thie/AppData/Local/atom/app-1.41.0/resources/app.asar/static/index.js:72:46)
    at require (C:\Users\Johnson.Thie\AppData\Local\atom\app-1.41.0\resources\app\static\<embedded>:11:146745)
    at bindings (C:\Users\Johnson.Thie\.atom\packages\pymakr\node_modules\bindings\bindings.js:81:44)
    at Object.<anonymous> (C:\Users\Johnson.Thie\.atom\packages\pymakr\node_modules\@serialport\bindings\lib\win32.js:1:191)
    at Object.<anonymous> (C:\Users\Johnson.Thie\.atom\packages\pymakr\node_modules\@serialport\bindings\lib\win32.js:111:3)
    at Module.get_Module._compile (C:\Users\Johnson.Thie\AppData\Local\atom\app-1.41.0\resources\app\static\<embedded>:11:147429)
    at Object.value [as .js] (C:\Users\Johnson.Thie\AppData\Local\atom\app-1.41.0\resources\app\static\<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:602:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
    at Function.Module._load (internal/modules/cjs/loader.js:533:3)
    at Module.require (file:///C:/Users/Johnson.Thie/AppData/Local/atom/app-1.41.0/resources/app.asar/static/index.js:72:46)
    at require (C:\Users\Johnson.Thie\AppData\Local\atom\app-1.41.0\resources\app\static\<embedded>:11:146745)
    at Object.<anonymous> (C:\Users\Johnson.Thie\.atom\packages\pymakr\node_modules\@serialport\bindings\lib\index.js:6:22)
    at Object.<anonymous> (C:\Users\Johnson.Thie\.atom\packages\pymakr\node_modules\@serialport\bindings\lib\index.js:17:3)
    at Module.get_Module._compile (C:\Users\Johnson.Thie\AppData\Local\atom\app-1.41.0\resources\app\static\<embedded>:11:147429)
    at Object.value [as .js] (C:\Users\Johnson.Thie\AppData\Local\atom\app-1.41.0\resources\app\static\<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:602:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
    at Function.Module._load (internal/modules/cjs/loader.js:533:3)
    at Module.require (file:///C:/Users/Johnson.Thie/AppData/Local/atom/app-1.41.0/resources/app.asar/static/index.js:72:46)
    at require (C:\Users\Johnson.Thie\AppData\Local\atom\app-1.41.0\resources\app\static\<embedded>:11:146745)
    at Object.<anonymous> (C:\Users\Johnson.Thie\.atom\packages\pymakr\node_modules\serialport\lib\index.js:2:17)
    at Object.<anonymous> (C:\Users\Johnson.Thie\.atom\packages\pymakr\node_modules\serialport\lib\index.js:17:3)
    at Module.get_Module._compile (C:\Users\Johnson.Thie\AppData\Local\atom\app-1.41.0\resources\app\static\<embedded>:11:147429)
The error was thrown from the pymakr package. You can help by creating an issue. Please explain what actions triggered this error.

## Any other comments?

None

## What versions of software are you using?

**Operating system:** Windows 10 64

**Atom version:** 1.41.0 x64

**Pymakr version:** 1.4.17

johnsonthie, 2019-10-24:
The readme advises to tell apm to use python 2. My apm is already using python 2.7.16 according to apm -v.

elmasria, 2019-10-24:
Hello @johnsonthie, Kindly note that we have a new version now 1.4.18 that will fix that issue. try to update the **pymakr** package

johnsonthie, 2019-10-24:
Awesome! The issue has been fixed. Thanks!

closed 2019-10-24 by johnsonthie