pycom/pymakr-atom #150: Uncaught Error: A dynamic link library (DLL) initialization routine failed./\?\$ATOM_HOME\packa...

ronnie14165, 2019-11-04:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.40.1 x64
**Electron**: 3.1.10
**OS**: Unknown Windows version
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.18

### Stack Trace

Uncaught Error: A dynamic link library (DLL) initialization routine failed.
\\?\C:\Users\14165\.atom\packages\pymakr\node_modules\@serialport\bindings\build\Release\bindings.node

```
At C:\Users\14165\.atom\packages\pymakr\node_modules\bindings\bindings.js:88

Error: A dynamic link library (DLL) initialization routine failed.
\\?\C:\Users\14165\.atom\packages\pymakr\node_modules\@serialport\bindings\build\Release\bindings.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:146745)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/win32.js:1:191)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:6:22)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/serialport/lib/index.js:2:17)
    at /packages/pymakr/node_modules/serialport/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:147429)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.4.18 
```

mc404, 2019-11-05:
I had the same error, but could work around it by:
1. uninstalling the pymakr plugin
2. deleting the folder C:\Users\[username]\.atom\.apm\pymakr\
3. reinstalling pymakr

Works fine again now.

chrisboer, 2019-11-09:
Same error but with @mc404 work around got it working again.

closed 2020-02-18 by pedroknup