pycom/pymakr-atom #151: Uncaught Error: error: 1114/\?\$ATOM_HOME\packages\pymakr\node_modules\@serialport\bindings\build...

NopMicrowave, 2019-11-08:
[Enter steps to reproduce:]

1. just after local update
2. on Atom start

**Atom**: 1.40.1 x64
**Electron**: 3.1.10
**OS**: Unknown Windows version
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.4.18

### Stack Trace

Uncaught Error: error: 1114\\?\C:\Users\nd216356\.atom\packages\pymakr\node_modules\@serialport\bindings\build\Release\bindings.node

```
At C:\Users\nd216356\.atom\packages\pymakr\node_modules\bindings\bindings.js:88

Error: error: 1114\\?\C:\Users\nd216356\.atom\packages\pymakr\node_modules\@serialport\bindings\build\Release\bindings.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:146745)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:81:44)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/win32.js:1:191)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/win32.js:111:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:6:22)
    at /packages/pymakr/node_modules/@serialport/bindings/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:147429)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:150977)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (/app.asar/static/index.js:72:46)
    at require (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:146745)
    at /packages/pymakr/node_modules/serialport/lib/index.js:2:17)
    at /packages/pymakr/node_modules/serialport/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.40.1/resources/app/static/<embedded>:11:147429)
```

### Commands

```
```

### Non-Core Packages

```
atom-flask 0.2.3 
atom-ide-debugger-node 0.7.3 
autocomplete-clang 0.12.0 
build 0.70.0 
busy-signal 2.0.1 
copy-with-style 1.1.0 
file-icons 2.1.35 
flask-snippets 0.3.3 
intentions 1.1.5 
language-ini 1.21.0 
language-mongodb 0.3.1 
linter 2.3.1 
linter-gcc 0.7.1 
linter-ui-default 1.7.1 
mongo 0.3.0 
node-debug 0.1.7 
plantuml-viewer 0.7.2 
platformio-ide 2.5.0 
platformio-ide-debugger 1.2.6 
platformio-ide-terminal 2.9.4 
pymakr 1.4.18 
remote-ftp 2.2.2 
tool-bar 1.1.12 
```

pedroknup, 2019-11-08:
I couldn't reproduce the issue. Have you tried removing the extension completely (%USERPROFILE%/.atom/packages/pymakr) and installing it again via Atom?

NopMicrowave, 2019-11-14:
Hi,

I have reinstall the package to version v1.4.18 and it works now.

closed 2019-11-14 by pedroknup