pycom/pymakr-atom #167: Uncaught TypeError: this.settings.sync_folder.replace is not a function

cmolitor, 2020-02-01:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.43.0 x64
**Electron**: 4.2.7
**OS**: Mac OS X 10.13.6
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.5.6

### Stack Trace

Uncaught TypeError: this.settings.sync_folder.replace is not a function

```
At /Users/christoph/.atom/packages/pymakr/lib/board/sync.js:125

TypeError: this.settings.sync_folder.replace is not a function
    at Sync.set_paths (/packages/pymakr/lib/board/sync.js:125:43)
    at /packages/pymakr/lib/board/sync.js:36:10)
    at Pymakr.sync (/packages/pymakr/lib/pymakr.js:658:22)
    at Pymakr.upload (/packages/pymakr/lib/pymakr.js:627:10)
    at /packages/pymakr/lib/pymakr.js:108:15)
    at PanelView.emit (events.js:182:13)
    at HTMLButtonElement.button_sync.onclick (/packages/pymakr/lib/main/panel-view.js:223:13)
```

### Commands

```
```

### Non-Core Packages

```
autocomplete-clang 0.12.1 
build 0.70.0 
busy-signal 2.0.1 
file-icons 2.1.36 
intentions 1.1.5 
language-ini 1.23.0 
linter 2.3.1 
linter-gcc2 0.8.7 
linter-ui-default 1.8.0 
platformio-ide 2.6.0 
platformio-ide-debugger 1.2.6 
platformio-ide-terminal 2.10.0 
pymakr 1.5.6 
tool-bar 1.2.0 
```

pedroknup, 2020-02-03:
Thank you for reporting. A new Pymakr version will be released soon with the fix for this bug.

cmolitor, 2020-02-07:
installed the new version but still have the issue

cmolitor, 2020-02-07:
seems to work now

closed 2020-02-07 by cmolitor