pycom/pymakr-atom #172: Uncaught TypeError: Cannot read property '77' of undefined

g0hww, 2020-02-10:
[Enter steps to reproduce:]

1. ... Specify the IPv4 address of a Lopy4 (that was previously reachable from a manual telnet session which is not interfering with the subsequent step) in the pymakr plugin settings and restart Atom.
2. ... Attempt to connect to a Device specified by an IPv4 address, resulting in a sequence of three Uncaught Type errors, for property 77, 255 and 76, the first of which is shown below.

**Atom**: 1.43.0 x64
**Electron**: 4.2.7
**OS**: Ubuntu 16.04.6
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 2.0.0

### Stack Trace

Uncaught TypeError: Cannot read property '77' of undefined

```
At /home/darren/.atom/packages/pymakr/lib/connections/telnet/util-telnet.js:87

TypeError: Cannot read property '77' of undefined
    at /packages/pymakr/lib/connections/telnet/util-telnet.js:87:23)
    at Socket._sock.on.data (/packages/pymakr/lib/connections/telnet/util-telnet.js:64:11)
    at Socket.emit (events.js:182:13)
    at addChunk (_stream_readable.js:283:12)
    at readableAddChunk (_stream_readable.js:264:11)
    at Socket.Readable.push (_stream_readable.js:219:10)
    at TCP.onStreamRead [as onread] (internal/stream_base_commons.js:94:17)
```

### Commands

```
```

### Non-Core Packages

```
auto-detect-indentation 1.3.0 
minimap 4.29.9 
pymakr 2.0.0 
python-indent 1.2.5 
symbols-list 2.5.4 
symbols-tree-view 0.14.0 
```

g0hww, 2020-02-10:
Note that if I leave an existing manual telnet connection open and then try to connect from Pymakr, I see this in the console (no error popups)

```
Connecting to 192.168.0.30...
Failed to connect (Connection refused).
```

cvillegas1976, 2020-02-12:
I have the same problem as reported.
I just installed version 2.0.1 with the same result.
Local connection via telnet is OK. Connection via serial port is OK.
Greetings.

pedroknup, 2020-02-13:
Thanks for reporting. I'm working on it.

pedroknup, 2020-02-13:
I've just created a PR with the fix for that problem. 
While we don't officially publish a patch release, please install Pymakr from that branch, manually. To do so, uninstall the current Pymakr, clone this repository, checkout the branch `telnet-77-255-bugfix` followed by `apm install && apm link`. 
Sorry for the inconvenience. 

You can expect an Pymakr update soon.

Thanks.

cvillegas1976, 2020-02-13:
thank you very much for your quick reply.
Unfortunately I do not see the branch indicated in the repository to apply the correction.
Thank you.

pedroknup, 2020-02-14:
My bad. I deleted it after merging it into develop.
Instead of to check that branch, please check develop branch instead.

pedroknup, 2020-02-18:
A new version (2.1.0) has been just released with a fix for that

cvillegas1976, 2020-02-18:
Installing “pymakr@2.1.0” failed.Hide output…
Failed to install pymakr because Git was not found.

The pymakr package has module dependencies that cannot be installed without Git.

You need to install Git and add it to your path environment variable in order to install this package.

You can install Git by downloading, installing, and launching GitHub for Windows: https://windows.github.com

Run apm -v after installing Git to see what version has been detected.

pedroknup, 2020-02-19:
Fixed. Please install Pymakr >2.1.1

closed 2020-02-19 by pedroknup