pycom/pymakr-atom #173: Uncaught TypeError: _this.pymakr.sync is not a function

fekher15, 2020-02-12:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.44.0 x64
**Electron**: 4.2.7
**OS**: Unknown Windows version
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 2.0.0

### Stack Trace

Uncaught TypeError: _this.pymakr.sync is not a function

```
At C:\Users\FSM\.atom\packages\pymakr\lib\main.js:48

TypeError: _this.pymakr.sync is not a function
    at HTMLElement.pymakrSync (/packages/pymakr/lib/main.js:48:47)
    at CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.44.0/resources/app/static/<embedded>:11:349290)
    at CommandRegistry.dispatch (~/AppData/Local/atom/app-1.44.0/resources/app/static/<embedded>:11:347765)
    at AtomEnvironment.dispatchApplicationMenuCommand (~/AppData/Local/atom/app-1.44.0/resources/app/static/<embedded>:1:764636)
    at EventEmitter.t (~/AppData/Local/atom/app-1.44.0/resources/app/static/<embedded>:1:772559)
    at EventEmitter.emit (events.js:182:13)
```

### Commands

```
     -6:51.9.0 application:reopen-project (atom-workspace.workspace.scrollbars-visible-always.theme-base16-tomorrow-light-theme.theme-one-light-ui)
     -6:38.4.0 application:open-folder (atom-pane.pane.active)
     -6:03.9.0 core:paste (input.hidden-input)
     -1:38.7.0 core:save (input.hidden-input)
  3x -1:04.5.0 editor:newline (input.hidden-input)
     -0:09.0 pymakr:run (button#pymakr-action-run.left-button)
     -0:03.2.0 pymakr:sync (button#pymakr-action-run.left-button)
```

### Non-Core Packages

```
atom-language-microalg 0.2.1 
Atom-Syntax-highlighting-for-Sass 0.5.0 
language-micro-cpp 0.1.2 
language-pymol 0.3.0 
microbit-python 0.1.6 
pymakr 2.0.0 
script 3.25.0 
teletype 0.13.4 
```

pedroknup, 2020-02-12:
Hi @fekher15. I couldn't reproduce your problem.

Have you tried reinstalling pymakr? You can try manually uninstalling Pymakr removing its folder under `%USERPROFILE%/.atom/packages` (Windows) or `~/.atom/packages` (macOS, linux) and then installing it again.

closed 2020-02-18 by pedroknup

gilgoldman, 2020-11-07:
Hi @pedroknup , this issue just surfaced when trying to sync a project with a target - reinstalling doesn't help.
Thoughts?