pycom/pymakr-atom #177: Pymakr 2.0.1 on Atom 1.44.0 install fail

ElectricJoe, 2020-02-12:
Following the install guide at [https://docs.pycom.io/pymakr/installation/atom/](https://docs.pycom.io/pymakr/installation/atom/) - by the way: it probably should be 'File > Settings' nowadays instead of 'Atom > Preferences > Install' under step 2. 

I can find the pymakr package and start install, however after a few 'WARN deprecated' messages I run into 'ERR! 404 Not Found: error-symbol@^0.1.0' and the process terminates.

Just wondering if that '^' character is a typo? Not sure where to fix this.

Log:
```
npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm ERR! code E404
npm ERR! 404 Not Found: error-symbol@^0.1.0
```

Operating system: Win10 x64
Atom version: 1.44.0 x64
Pymakr version: 2.0.1

Thanks in advance!

pedroknup, 2020-02-13:
Hi @ElectricJoe. Thanks for reporting. 
I'm working on it. I'll write you back as soon as I have found something. 
Pymakr doesn't have error-symbol library on its dependencies. But it does have some dependencies that it does. Have you tried making a clean installation (removing any previous installed Pymakr)? You can try deleting Pymakr folder under `%USERPROFILE%/.atom/packages` and try again.

ElectricJoe, 2020-02-13:
Pedro, that folder was empty except one generic readme.md file. Restarted the install via File > Settings > Install and it seemed to run through this time - without doing anything obviously different from previous attempts. I don't know what to say.

I've attached logs from previous failed attempts:
[2020-02-12T16_28_40_721Z-debug.log](https://github.com/pycom/pymakr-atom/files/4199215/2020-02-12T16_28_40_721Z-debug.log)
[2020-02-12T16_38_41_444Z-debug.log](https://github.com/pycom/pymakr-atom/files/4199216/2020-02-12T16_38_41_444Z-debug.log)

But now system seems to work. Thanks.

closed 2020-02-14 by pedroknup