pycom/pymakr-atom #180: second atom instance doesn't find the devices

doniks, 2020-02-14:
## What are the steps to reproduce this issue?
1. open first atom -> shows all connected devices
2. open second atom -> doesn't show any device
3. close first, 
4. close second
5. reopen -> now it's ok 

## What happens?
second atom instance doesn't find the devices

## What were you expecting to happen?
any open pymakr-atom should see the list of all devices

## Any logs, error output, etc?
I'll attach a screencapture

## Any other comments?

## What versions of software are you using?

**Operating system:**
Ubuntu 19.10

**Atom version:**
1.44 (snap)

**Pymakr version:**
2.0.1

doniks, 2020-02-14:
[second atom can't find devices.zip](https://github.com/pycom/pymakr-atom/files/4203920/second.atom.can.t.find.devices.zip)

pedroknup, 2020-02-14:
Hi @doniks. Thanks for reporting. 
A PR ([pb-893-toggle-terminal-fix](https://github.com/pycom/pymakr-atom/pull/181)) has just been created with the fix for that problem. 
While we don't officially publish a patch release, please install Pymakr from that branch, manually. To do so, uninstall the current Pymakr, clone this repository, checkout the branch `pb-893-toggle-terminal-fix` followed by `apm install && apm link`.
If that branch no longer exists, it means it has been merged into `develop`. In that case, checkout `develop` branch instead.
Sorry for the inconvenience.

You can expect a Pymakr update soon.

pedroknup, 2020-02-18:
A new version (2.1.0) has been just released with a fix for that

closed 2020-02-18 by pedroknup