pycom/pymakr-atom #183: Uncaught TypeError: _this.pyboard.disconnect_silent is not a function

cvillegas1976, 2020-02-14:
To reproduce
1. Telnet connect to IPv4 LoPy4
2. Download buton
3. All files are apparently downloaded correctly, however at the end of the connection it drops.

**Atom**: 1.44.0 x64
**Electron**: 4.2.7
**OS**: Unknown Windows version
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 2.0.1

Stack Trace

Uncaught TypeError: _this.pyboard.disconnect_silent is not a function

At C:\Users\Claudio\.atom\packages\pymakr\lib\board\shell.js:533

TypeError: _this.pyboard.disconnect_silent is not a function
    at finish (/packages/pymakr/lib/board/shell.js:533:23)
    at /packages/pymakr/lib/connections/pytelnet.js:95:15
    at Telnet.write (/packages/pymakr/lib/connections/telnet/util-telnet.js:192:5)
    at TelnetClient.write (/packages/pymakr/lib/connections/telnet/telnetcli.js:211:14)
    at PyTelnet.send_raw (/packages/pymakr/lib/connections/pytelnet.js:94:17)
    at PyTelnet.send (/packages/pymakr/lib/connections/pytelnet.js:90:10)
    at Pyboard.send (/packages/pymakr/lib/board/pyboard.js:527:21)
    at Pyboard.softResetNoFollow (/packages/pymakr/lib/board/pyboard.js:201:10)
    at /packages/pymakr/lib/board/shell.js:374:21
    at /packages/pymakr/lib/board/pyboard.js:652:9
    at /packages/pymakr/lib/connections/pytelnet.js:95:15
    at Telnet.write (/packages/pymakr/lib/connections/telnet/util-telnet.js:192:5)
    at TelnetClient.write (/packages/pymakr/lib/connections/telnet/telnetcli.js:211:14)
    at PyTelnet.send_raw (/packages/pymakr/lib/connections/pytelnet.js:94:17)
    at Pyboard.sendRaw (/packages/pymakr/lib/board/pyboard.js:644:21)
    at Pyboard.execRawNoReset (/packages/pymakr/lib/board/pyboard.js:650:10)
    at Shell.reset (/packages/pymakr/lib/board/shell.js:372:18)
    at Shell.cleanClose (/packages/pymakr/lib/board/shell.js:545:12)
    at /packages/pymakr/lib/board/shell.js:494:13
    at Shell.stopWorking (/packages/pymakr/lib/board/shell.js:522:7)
    at Shell.exit (/packages/pymakr/lib/board/shell.js:493:10)
    at Sync.exit (/packages/pymakr/lib/features/sync/sync.js:825:16)
    at lcb (/packages/pymakr/lib/features/sync/sync.js:615:13)
    at /packages/pymakr/lib/features/sync/sync.js:632:11
    at /packages/pymakr/lib/helpers/utils.js:132:7
    at next (/packages/ftp-remote-edit/node_modules/rimraf/rimraf.js:83:7)
    at CB (/packages/ftp-remote-edit/node_modules/rimraf/rimraf.js:119:9)
    at /packages/ftp-remote-edit/node_modules/rimraf/rimraf.js:145:14
    at FSReqWrap.oncomplete (fs.js:154:21)

Commands

     -1:36.7.0 application:add-project-folder (atom-pane.pane.active)

Non-Core Packages

autocomplete-clang 0.12.1 
build 0.70.0 
busy-signal 2.0.1 
file-icons 2.1.39 
ftp-remote-edit 0.18.0 
intentions 1.1.5 
language-ini 1.23.0 
linter 2.3.1 
linter-gcc 0.9.0 
linter-gcc2 0.8.8 
linter-ui-default 1.8.1 
pymakr 2.0.1 
tool-bar 1.2.0

pedroknup, 2020-02-18:
Hi @cvillegas1976. Thanks for reporting. A new Pymakr version will be released very soon with the fix for that. I'll write you back once it's done.

pedroknup, 2020-02-18:
A new version (2.1.0) has been just released with a fix for that

closed 2020-02-18 by pedroknup