pycom/pymakr-atom #208: Feature "Get device info" does not seem to work

JacobJustCoding, 2020-05-19:
## What are the steps to reproduce this issue?
1. Connect the WiPy board via USB port.
2. Click on "Get device info"

## What happens?
Actually, the REPL seems to get stuck. The cursor blinks but no information has been shown.

## What were you expecting to happen?
I assume that the purpose of this feature is to display information about the current firmware version and other additional information

## What versions of software are you using?
Pycom MicroPython 1.20.2.rc3 [v1.11-2037465] on 2020-02-05; WiPy with ESP32
Pybytes Version: 1.3.1

**Operating system:**
Windows 10

**Atom version:**
Atom version: 1.46.0

**Pymakr version:**
2.1.3

pedroknup, 2020-05-19:
Hi @JacobJustCoding. Thanks for reporting. 
I couldn't reproduce that problem with a WiPy. Could you please check the console inside of developer tools to see if there is any errors?

JacobJustCoding, 2020-05-19:
Hello @pedroknup,
first of all, I'd like to thank you for your quick answer.
I'm not really good at Javascript so I'm sending a screenshot from console after clicking "Get device info" button.
I will be very glad if you could give me not only a solution but also an explanation what is the reason of this bug.
[image]

elmasria, 2021-05-17:
Please check the latest version of Pymakr.

closed 2021-05-17 by elmasria