pycom/pymakr-atom #209: Uncaught NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not ...

peter-pycom, 2020-05-19:
[Enter steps to reproduce:]

1. open Atom
2. open /dev/ttyACM0
3. close /dev/ttyACM0 with the little X 

-> 

REPL becomes empty
/dev/ttyACM0 stays with a green dot
another tab /dev/ttyACM0 with a red dot opens next to it 
and exception shows

**Atom**: 1.46.0 x64
**Electron**: 4.2.7
**OS**: Ubuntu 20.04
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 2.1.3

### Stack Trace

Uncaught NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.

```
At /home/peter/.atom/packages/pymakr/node_modules/xterm/lib/xterm.js:1

Error: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.
    at t.dispose (/packages/pymakr/node_modules/xterm/lib/xterm.js:1:215082)
    at t.dispose (/packages/pymakr/node_modules/xterm/lib/xterm.js:1:234675)
    at t.dispose (/packages/pymakr/node_modules/xterm/lib/xterm.js:1:107782)
    at e.dispose (/packages/pymakr/node_modules/xterm/lib/xterm.js:1:99263)
    at Device.close (/packages/pymakr/lib/connections/device.js:195:25)
    at Pymakr.removeDevice (/packages/pymakr/lib/pymakr.js:501:19)
    at /packages/pymakr/lib/pymakr.js:346:13)
    at PanelView.emit (events.js:182:13)
    at /packages/pymakr/lib/views/panel-view.js:512:13)
    at HTMLDivElement.dispatch (/packages/pymakr/node_modules/jquery/dist/jquery.js:5429:27)
    at HTMLDivElement.elemData.handle (/packages/pymakr/node_modules/jquery/dist/jquery.js:5233:28)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 2.1.3 
```

stsdc, 2020-05-19:
Just update to 2.1.3 and have this error

pedroknup, 2020-05-20:
Thank you for reporting. I'm working on it.

pedroknup, 2020-05-20:
This problem has been fixed with today's update of Pymakr (2.1.4). Please update your Pymakr Atom :)

peter-pycom, 2020-05-20:
Awesome! I confirm it works in 2.1.4

closed 2020-05-20 by peter-pycom