pycom/pymakr-atom #222: Atom portable version - Console doesn't appear

clavay, 2020-10-13:
## What are the steps to reproduce this issue?
1. Download portable version of Atom : https://flight-manual.atom.io/getting-started/sections/installing-atom/#portable-mode
2. create .atom directory and Install Pymakr
3. Use it in a new computer or copy the folder with atom and .atom in a new place

## What happens?
The pymakr console doesn't appear after moving the directory with atom and .atom or after starting atom on a new computer (when it is on a USB)

## What were you expecting to happen?
the pymakr console continue to appear

## Any logs, error output, etc?
No

## Any other comments?
Sometime reinstalling pymakr solve the problem, sometime no. I don't know why.

**Operating system:**
Windows 10
**Atom version:**
1.51.0
**Pymakr version:**
2.1.5

William04A, 2020-10-14:
I just reinstalled PyMakr and Atom and the PyMakr plugin does no longer appear in the bottom right corner.

I assume we have the same issue, even though I am not running a portable version of Atom.

William04A, 2020-10-14:
I downgraded the version of Atom and now it works...

clavay, 2020-10-14:
Yes it seems the new 1.52.0 version of atom doesn't work with pymakr plugin.

William04A, 2020-10-14:
I used an installer release file from ‎8th September 2020. That version worked.

elmasria, 2020-10-14:
@clavay please check the new Pymakr version. it solves the broken part in the new atom version

clavay, 2020-10-16:
it works with the last release

closed 2020-10-16 by clavay