pycom/pymakr-atom #238: Uncaught TypeError: atom.confirm(...).then is not a function

jthompson-ceres, 2021-01-15:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.54.0 x64
**Electron**: 6.1.12
**OS**: Mac OS X 10.15.7
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 2.1.8

### Stack Trace

Uncaught TypeError: atom.confirm(...).then is not a function

```
At /Users/joshua/.atom/packages/pymakr/lib/features/sync/sync.js:386

TypeError: atom.confirm(...).then is not a function
    at /packages/pymakr/lib/features/sync/sync.js:386:12
```

### Commands

```
```

### Non-Core Packages

```
pymakr 2.1.8 
```

elmasria, 2021-01-15:
Hello @jthompson-ceres,

Please note that a new version is available that solves the above issue.

JoshuaBThompson, 2021-04-24:
@elmasria still happening on atom 1.55.0 x64 (mac osx)

closed 2021-05-17 by elmasria