pycom/pymakr-atom #244: Can't install on fresh ATOM / Pymakr install, throws error

ericarnoldy, 2021-04-14:
I can't seem to install on ATOM 1.56.0 and pymakr 2.1.9, I get the following error:

> @serialport/bindings@8.0.8 install C:\Users\Admin\AppData\Local\Temp\apm-install-dir-2021314-3776-nbat22.l5noj\node_modules\pymakr\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

C:\Users\Admin\AppData\Local\Temp\apm-install-dir-2021314-3776-nbat22.l5noj\node_modules\pymakr\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\Users\Admin\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\Admin\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
prebuild-install WARN install No prebuilt binaries found (target=9.4.4 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! find VS 
gyp ERR! find VS msvs_version not set from command line or npm config
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS could not use PowerShell to find Visual Studio 2017 or newer
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS 
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS 
gyp ERR! configure error 
gyp ERR! stack Error: Could not find any Visual Studio installation to use
gyp ERR! stack     at VisualStudioFinder.fail (C:\Users\Admin\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:121:47)
gyp ERR! stack     at C:\Users\Admin\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:74:16
gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\Users\Admin\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:351:14)
gyp ERR! stack     at C:\Users\Admin\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:70:14
gyp ERR! stack     at C:\Users\Admin\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:372:16
gyp ERR! stack     at C:\Users\Admin\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7
gyp ERR! stack     at C:\Users\Admin\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:297:5)
gyp ERR! stack     at ChildProcess.emit (events.js:200:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:1021:16)
gyp ERR! System Windows_NT 10.0.19041
gyp ERR! command "C:\\Users\\Admin\\AppData\\Local\\atom\\app-1.56.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Admin\\AppData\\Local\\atom\\app-1.56.0\\resources\\app\\apm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Admin\AppData\Local\Temp\apm-install-dir-2021314-3776-nbat22.l5noj\node_modules\pymakr\node_modules\@serialport\bindings
gyp ERR! node -v v12.4.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN notsup Unsupported engine for pymakr@2.1.9: wanted: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current: {"node":"12.4.0","npm":"6.14.8"})
npm WARN notsup Not compatible with your version of node/npm: pymakr@2.1.9
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\Admin\AppData\Local\Temp\apm-install-dir-2021314-3776-nbat22.l5noj\package.json'
npm WARN apm-install-dir-2021314-3776-nbat22.l5noj No description
npm WARN apm-install-dir-2021314-3776-nbat22.l5noj No repository field.
npm WARN apm-install-dir-2021314-3776-nbat22.l5noj No README data
npm WARN apm-install-dir-2021314-3776-nbat22.l5noj No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Admin\.atom\.apm\_logs\2021-04-14T16_00_03_466Z-debug.log

elmasria, 2021-04-15:
Hi @ericarnoldy, we are trying to fix this issue

ericarnoldy, 2021-04-16:
@elmasria Thank you!  Is there even a manual workaround we could try, we're kind of dead in the water at this point

elmasria, 2021-04-17:
@ericarnoldy, one option could be is to downgrade **Atom** which will be compatible with Pymakr

elmasria, 2021-04-17:
@ericarnoldy a new version is now available 2.1.10 please update and check. let me know if you face any issue

teuler, 2021-04-17:
Also version 2.1.10 does not work with latest Atom. Crashes Atom during the installation.

elmasria, 2021-04-17:
@teuler thanks for this feedback which windows you have 64 or 32?

teuler, 2021-04-17:
Win 10 64

elmasria, 2021-04-18:
@teuler please check version 2.1.11

teuler, 2021-04-18:
All Windows 10 (64 bit):
- Atom 1.55.0 + pymakr 2.1.11 works
- Atom 1.56.0 + pymakr 2.1.11 works
Thanks @elmasria

closed 2021-05-05 by elmasria