pycom/pymakr-atom #245: Pymakr 2.1.9 not working with Atom 1.56 on Macos Big Sur

dirkpesch, 2021-04-15:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1.
2.
3.

## What happens?
Pymakr 2.1.9 does not start a REPL console on the latest version of Atom 1.56 on MacOS Big Sur. It works with Atom 1.55 though

## What were you expecting to happen?

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## What versions of software are you using?

**Operating system:**

**Atom version:**

**Pymakr version:**

IoHL, 2021-04-16:
I got the same issue also. Here is the logs:

```
Error while loading serialport library
/Users/iohl/.atom/packages/pymakr/lib/main.js:105 Error: Could not locate the bindings file. Tried:
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/build/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/build/Release/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/Debug/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/out/Release/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/Release/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/build/default/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/compiled/12.14.1/darwin/x64/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/binding/node-v80-darwin-x64/bindings.node
    at bindings (/Users/iohl/.atom/packages/pymakr/node_modules/bindings/bindings.js:126)
    at Object.<anonymous> (/Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/darwin.js:2)
    at Object.<anonymous> (/Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/darwin.js:112)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11)
    at Module.load (internal/modules/cjs/loader.js:815)
    at Module._load (internal/modules/cjs/loader.js:727)
    at Function.Module._load (electron/js2c/asar.js:769)
    at Module.require (index.js:72)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11)
    at Object.<anonymous> (/Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/index.js:10)
    at Object.<anonymous> (/Users/iohl/.atom/packages/pymakr/node_modules/@serialport/bindings/lib/index.js:17)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11)
    at Module.load (internal/modules/cjs/loader.js:815)
    at Module._load (internal/modules/cjs/loader.js:727)
    at Function.Module._load (electron/js2c/asar.js:769)
    at Module.require (index.js:72)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11)
    at Object.<anonymous> (/Users/iohl/.atom/packages/pymakr/node_modules/serialport/lib/index.js:2)
    at Object.<anonymous> (/Users/iohl/.atom/packages/pymakr/node_modules/serialport/lib/index.js:17)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11)
    at Module.load (internal/modules/cjs/loader.js:815)
    at Module._load (internal/modules/cjs/loader.js:727)
    at Function.Module._load (electron/js2c/asar.js:769)
    at Module.require (index.js:72)
    at require (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11)
    at Object.prepareSerialPort (/Users/iohl/.atom/packages/pymakr/lib/main.js:87)
    at Object.activate (/Users/iohl/.atom/packages/pymakr/lib/main.js:13)
```

elmasria, 2021-04-17:
@dirkpesch Please check the new version 2.1.10

closed 2021-05-05 by elmasria