pycom/pymakr-atom #248: Mac Atom PyMakr (2.1.12) not downloading

grahamPatico, 2021-04-19:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Downloading pymakr from the atom package list
2.
3.

## What happens?
I cannot download pymakr from atom.

## What were you expecting to happen?
I expected to download the package from Atom but pymakr failing. I tried to get a fresh download of atom and still failed.

## Any logs, error output, etc?

'> @serialport/bindings@8.0.8 install /private/var/folders/8c/qcltmkyd0ql993vylmy1y1jr0000gn/T/apm-install-dir-2021318-4304-p5tdlw.20yk9/node_modules/pymakr/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
prebuild-install WARN install No prebuilt binaries found (target=9.4.4 runtime=electron arch=x64 libc= platform=darwin)
No receipt for 'com.apple.pkg.CLTools_Executables' found at '/'.

No receipt for 'com.apple.pkg.DeveloperToolsCLILeo' found at '/'.

No receipt for 'com.apple.pkg.DeveloperToolsCLI' found at '/'.

gyp: No Xcode or CLT version detected!
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:200:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Darwin 20.3.0
gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /private/var/folders/8c/qcltmkyd0ql993vylmy1y1jr0000gn/T/apm-install-dir-2021318-4304-p5tdlw.20yk9/node_modules/pymakr/node_modules/@serialport/bindings
gyp ERR! node -v v12.4.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN notsup Unsupported engine for pymakr@2.1.12: wanted: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current: {"node":"12.4.0","npm":"6.14.8"})
npm WARN notsup Not compatible with your version of node/npm: pymakr@2.1.12
npm WARN enoent ENOENT: no such file or directory, open '/private/var/folders/8c/qcltmkyd0ql993vylmy1y1jr0000gn/T/apm-install-dir-2021318-4304-p5tdlw.20yk9/package.json'
npm WARN apm-install-dir-2021318-4304-p5tdlw.20yk9 No description
npm WARN apm-install-dir-2021318-4304-p5tdlw.20yk9 No repository field.
npm WARN apm-install-dir-2021318-4304-p5tdlw.20yk9 No README data
npm WARN apm-install-dir-2021318-4304-p5tdlw.20yk9 No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/grahampatterson/.atom/.apm/_logs/2021-04-19T02_04_21_542Z-debug.log
'

## Any other comments?
f

## What versions of the software are you using?

**Operating system:**
Big Sur 11.2.1
**Atom version:**
1.56.0 x64
**Pymakr version:**
2.1.12

elmasria, 2021-04-19:
Hello @grahamPatico, can you remove the old Pymakr, and clear the packages under `~/.atom` (or maybe remove the entire ~/.atom) folder and try to install again. I am checking from my side and working fine.

grahamPatico, 2021-04-19:
I finally got it working with Atom 1.54.0. I couldn't pymakr working on Atom ever after deleting .atom files and deleting the application on my mac until I downgrades the Atom version from 1.56.0.

teuler, 2021-04-19:
Works for me for Atom 1.56.0 (64bit; Windows 10 64bit), although not at first.
Update hangs. Restarting Atom, deinstalling pymakr, restarting Atom, freshly installing pymakr worked for me. The installation still takes unusually long.

closed 2021-05-05 by elmasria

mapj1984, 2021-05-24:
Hey there! happen the same, I downloaded the latest version, and I had to downgrade the version to Atom 1.54.0.

elmasria, 2021-05-25:
Hi @mapj1984,

Thank you for your feedback. 

I think this issue is also related to the rebuilding of the serial port (we still using the old version) and same as with Windows OS you have to have some extra packages (software) that will helps doing the rebuild on installation. 

Please make sure you have (Xcode or CLT) as I can see from the error message (No Xcode or CLT version detected)

Helpful links: https://medium.com/flawless-app-stories/gyp-no-xcode-or-clt-version-detected-macos-catalina-anansewaa-38b536389e8d

Let me know if you have other questions or you face a different issue. 

Best Regards,
Ahmad EL Masri